esphome-econet/esphome-econet #46: Refactor requesting after writing.

stockmopar, 2023-09-13:
This could cause very close back to back requests being made. This also bypasses the logic of trying to pick a good time to make a request. Since we are making request every 500ms this is probably not needed. But if periodic requests are extended it would be to make this request as soon as its a good time to make the request.

https://github.com/esphome-econet/esphome-econet/blob/1f98c686b6a146fbb2da36617b2ec9c3a3538fcf/components/econet/econet.cpp#L484

tronikos, 2023-09-13:
I've already done this locally as part of #42. Need to do some more testing before I send a PR. I've also changed the logic of trying to pick a good time and it doesn't seem I get any more messages with crc16 mismatches.

stockmopar, 2023-09-13:
What logic are you proposing to change to pick a good time?

tronikos, 2023-09-13:
I'll add you as reviewer. The main issue of the current implementation is it can send a message in the middle of receiving a message. With my changes it will wait for partially received message to be fully received or timed out.

PR #49 (2023-09-14): Add update_interval support defaulting to 30s and rework the logic when it's ready to send messages
Closes #42 
Closes #46

Please review #47 first and then the last 3 commits here.
closed 2023-09-14 by tronikos