esphome-econet/esphome-econet #61: Provide Better Handling of Alerts and Populate Specific Alert Information

barndawgie, 2023-09-15:
Currently, our configurations  don't include any details on Alerts. We need to get both Alert State (i.e. is there an active alert) and specific details on the active alert into the basic configurations.

This issue is blocked on the moment on getting an actual Alert State on a device so we can see how the ALRMALRT and ALMCODE1 states are populated.

tronikos, 2023-09-28:
I was able to get 2 alerts by disconnecting the temperature sensors from the elements. I initially disconnected just the lower one and the value I got was 1. Unfortunately I didn't get any value for ALMCODE1 or the rest in the ESPHome logs in the responses of the read requests. Also, not sure why but the binary sensor stayed OK the whole time. The count sensor increased to 2.

This was my config:

```yaml
binary_sensor:
  - platform: econet
    name: "Active Alerts"
    sensor_datapoint: ALRMALRT
    device_class: "problem"

sensor:
  - platform: econet
    name: "Active Alerts"
    sensor_datapoint: ALRMALRT
    accuracy_decimals: 0
    icon: "mdi:water-boiler-alert"
    state_class: "measurement"
    entity_category: "diagnostic"
  - platform: econet
    name: "Alert 1"
    sensor_datapoint: ALMCODE1
    icon: "mdi:water-boiler-alert"
    state_class: "measurement"
    entity_category: "diagnostic"

text_sensor:
  - platform: econet
    name: "Alert 1"
    sensor_datapoint: ALMCODE1
    icon: "mdi:water-boiler-alert"
    entity_category: "diagnostic"
```

This is the history in Home Assistant:
[image]

So let's only add the following:

```yaml
sensor:
  - platform: econet
    name: "Active Alerts"
    sensor_datapoint: ALRMALRT
    accuracy_decimals: 0
    icon: "mdi:water-boiler-alert"
    state_class: "measurement"
    entity_category: "diagnostic"
```

barndawgie, 2023-09-28:
> device_class probably has something to do with why it stayed `OK`

It shouldn't be. I think Device Class just affects icon and naming (i.e. Problem/OK vs. On/Off). It should still be 0 = Off and 1 = On. We thought 2 would mean the same as 1 but perhaps that's not the case...

tronikos, 2023-10-02:
I was able to get alert text via ALARM_01

PR #86 (2023-09-28): Add Active Alerts sensor and bump version to 1.0.0
Fixes #61
closed 2023-10-03 by barndawgie