esphome-econet/esphome-econet #74: Automated Builds are Using Remote Rather than Local Source

barndawgie, 2023-09-23:


tronikos, 2023-10-15:
We are still using remote yaml instead of local.

https://github.com/esphome-econet/esphome-econet/pull/109 failed to build see https://github.com/esphome-econet/esphome-econet/actions/runs/6522996217/job/17713165299?pr=109 with:
```
INFO Reading configuration example-esp32.yaml...
INFO Cloning https://github.com/esphome-econet/esphome-econet@main
Failed config

econet: [source .esphome/packages/e86e50df/econet_base.yaml:53]
  
  'src_address' is a required option for [econet].
  uart_id: uart_0
  update_interval: 30s
  
  [model] is an invalid option for [econet]. Please check the indentation.
  model: Heatpump
Error: Process completed with exit code 2.
```

Even locally it fails:

```
$ esphome -s external_components_source components compile example-esp32.yaml
INFO ESPHome 2023.9.3
INFO Reading configuration example-esp32.yaml...
INFO Cloning https://github.com/esphome-econet/esphome-econet@main
Failed config

econet: [source .esphome\packages\e86e50df\econet_base.yaml:53]

  'src_address' is a required option for [econet].
  uart_id: uart_0
  update_interval: 30s

  [model] is an invalid option for [econet]. Please check the indentation.
  model: Heatpump
```

With `-s external_components_source` we only get to use the local components directory. 

Because of:
```yaml
packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    # Change this file to the correct one for your device
    file: econet_heatpump_water_heater.yaml
```

we use the remote yaml package.

barndawgie, 2023-10-16:
Looks like this is an issue with the example-* configs, right? I missed that usecase and it is using the remote config (even though it's getting the right code).

Given the issue I filed with esphome (https://github.com/esphome/issues/issues/4939) I'm not sure how to fix this. Let me do some research and see what I can figure out.

barndawgie, 2023-10-21:
This was fixed by #112 though we may want to re-enable config validation in the future if there is a straightforward way to doit.

closed 2023-10-21 by barndawgie