esphome-econet/esphome-econet #120: Any chance of supporting gas water heaters? REWRA631GWH

thenoid, 2023-10-27:
https://parts.rheem.com/product/RPD-EEWRA631GWH

Have a gas water heater, econet doesn't have a RJ11 instead just a 3 wire connection, guessing power+ground+data/1wire.

thenoid, 2023-10-28:
Aye, it plugs into the gas control module:
https://youtu.be/jKL5fwGmSaM?si=UYk_HSESeqsezlTQ&t=139
https://www.homedepot.com/p/Rheem-EcoNet-Home-Comfort-Wi-Fi-Module-for-Performance-Platinum-Gas-Smart-Tank-Water-Heaters-REWRA631GWH/205310140

thenoid, 2024-01-17:
Howdy again - snagged on the required hardware, what can I do to help get this going/add support?

Hasstuff, 2024-01-17:
I too am interested in this possibility. I am currently using the HACS custom Home Assistant Rheem Econet integration with the following:.

WV8840 gas valve 
https://www.venturii.net/blog/2015/09/07/honeywell-wv8840a-gas-valve/
connected through a
 " Bradford White ICON 239-47872-00 Accessory Module Kit" 
https://www.bradfordwhiteparts.com/product/bradford-white-239-47872-00-plug-in-module
To a WiFi module.
https://parts.rheem.com/product/RPD-EEWRA631GWH
Assembled using this for reference:
https://forums.raspberrypi.com/viewtopic.php?t=136314

This works most of the time however the WiFi module appears to be the week link. The module will not reconnect to WiFi reliably. Manually rebooting is often required.

Although my gas valve is powered through a thermopile rather than powered by 24 volts ( as appears to be the case with @thenoid gas valve), they both can connect to the same  " Bradford White ICON 239-47872-00 Accessory Module Kit" .
https://www.manualslib.com/manual/1911540/Bradford-White-Accessory-Module.html?page=12#manual
Neither the  Bradford White Accessory Kit or the RPD-EEWRA631GWH WiFi module have an RJ11 port. 

I have the recommended hardware to begin testing.
https://github.com/esphome-econet/esphome-econet/wiki/Recommended-Hardware-Purchase-and-Setup-Instructions

I am willing to assist as I would prefer local control.

thenoid, 2024-01-18:
[image]

WV4262E1107 appears to be the model of the gas valve included on my rheem heater.

DaveStLou, 2024-03-23:
I'm also interested in this functionality. I have the hardware installed and willing to test.

DaveStLou, 2024-07-01:
Is there some testing we can do to determine how much modification is needed and help get this enhancement scoped out?

DaveStLou, 2024-07-20:
I don't know how to do that. If you can give me instructions or point me to instructions, I'll be glad to help.

dtylerlivingston, 2025-01-14:
I too have a Rheem Gas Water Heater and would like to integrate with ESPHome.  As [thenoid](https://github.com/thenoid) mentioned, it does not have an RJ11 connector.  I am not sure the type of connector (screen shot below).  It does have yellow, black, and red wires going to it from the wifi module.  

Rheem 50gal Natural Gas Water Heater
https://www.homedepot.com/p/Rheem-Performance-Platinum-50-Gal-Tall-40-000-BTU-Powered-Damper-Natural-Gas-Water-Heater-with-12-Year-Warranty-XG50T12DM40U0/204318411#overlay

Rheem® Home Comfort WiFi Module
https://images.thdstatic.com/catalog/pdfImages/89/89c3dd15-290b-4b70-b6d4-f1d124039a6f.pdf

The valve on the water heater is a Resideo WV4262E1107 which has a 3-pin plug that the wifi module plugs into.  
https://www.suppliesdepot.com/mm5/graphics/pdf/RHEEM%20SP20296%20GAS%20CONTROL%20SERVICE%20PART.pdf

[image]

dtylerlivingston, 2025-01-14:
After googling some more tonight, I think this is a 3-pin Molex port, specifically Honeywell's EnviraCOM™ Remote Communication Port.  The wifi module I am using references the trademark to Honeywell EnviraCOM Protocol 1.1.  However there appears to be several different types of products from thermostats to water heaters that use this serial communication protocol.

https://www.wink.com/downloads/help/rheem-econet-water-heaters/wifi-module-installation-instructions.pdf
[image]

https://github.com/roy-bentley/enviracom/blob/main/docs/reference/W8735A1005_SPECS.pdf
[image]

https://forums.homeseer.com/forum/homeseer-products-services/general-discussion-area/54182-honeywell-enviracom-zone-controllers/page8

https://www.bloominglabs.org/Water_Heater_Monitoring_and_Setback_Control

This board for an automation platform called CommandFusion, seems to have C code (open at own risk).  However it mentions dampeners so it appears to be HVAC related.  
https://groups.google.com/g/commandfusion/c/J0jmAoqP_YY?pli=1
https://groups.google.com/group/commandfusion/attach/d208eb611734e808/Enviracom%20Pronto.zip?part=0.1&view=1

thezfunk, 2025-01-15:
My guess is we need to hook up the RS232 to the ESP32 and then just listen with software of some kind?  Figure out how it is sending temp and what happens when you call for heat, etc.  Some of them might be the same as from the electric models.  I can physically hook it up by I don't know what software we would need to listen.

thenoid, 2025-01-15:
In the discord I supplied images of the internal chipsets of both the valve
and the control unit.  Things kind of died off though, since it's EnviraCom.

dtylerlivingston, 2025-01-21:
I have an AtomS3 Lite with RS485 module.  Could this be used with Arduino IDE Serial Monitor?  The water heater itself provides the power to the wifi module.  The wires of the wifi module are yellow, red, and black which I assume would align with A (green), 12v DC (red) , and ground (black) on RS485.  However, I am concerned as the RS485 is 5 to 12v DC and I think the wifi module is 24v AC.  If I only want to get data, would I just use the yellow into the RS485?

barndawgie, 2025-01-21:
> I have an AtomS3 Lite with RS485 module.  Could this be used with Arduino IDE Serial Monitor?  The water heater itself provides the power to the wifi module.  The wires of the wifi module are [...]

If you don't want power just don't hook up the 5V/12V/24V line. You still need A,B, and Ground.

I'm not sure about software for reverse engineering the RD485 - I think @stockmopar is the expert there. I recommend posting in the Discord for help with that aspect.

sarabveer, 2026-10-16:
Posting this here since this issue is one of the first search results for this module.

I bought a Rheem REWRA631GWH, and it came without the wiring harness. I got it connected to the Resideo WV4464B1037 gas control on my A.O. Smith 50 Gal Power Vent.

With the front of the Wi-Fi module facing up, I’m numbering the connector pins **1–6 from left to right**. My gas control’s 3-pin EnviraCOM port uses the same order as the first three pins:

| Pin | Harness color | Purpose |
|---|---|---|
| 1 | Black | 24VAC / R |
| 2 | Red | Common / C |
| 3 | Yellow | Data |

Ignore pins 4–6 for this setup: pin 4 is unused, and pins 5–6 are for the leak sensor, which I don’t have.

I connected pin 1 to pin 1, pin 2 to pin 2, and pin 3 to pin 3.

My gas controller didn’t supply 24VAC. Both the gas controller’s EnviraCOM connection and the Wi-Fi module needed 24VAC, so I connected an external **24VAC transformer across pins 1 and 2 on both**. Communication then worked.

tankdeer, 2026-10-16:
@sarabveer Maybe I am misreading, but this is for you connecting the REWRA631GWH to your water heater. I don't see any mention of connecting to an ESP device for monitoring via ESPHome.

sarabveer, 2026-10-16:
> @sarabveer Maybe I am misreading, but this is for you connecting the REWRA631GWH to your water heater. I don't see any mention of connecting to an ESP device for monitoring via ESPHome. 

Right, the gas controller uses EnviraCOM.

So if any ESPHome solution would be developed it would need to run using 24VAC and talk the EviraCOM protocol which so far seems like a dead end.

Might be easier to somehow local control the REWRA631GWH module.