esphome-econet/esphome-econet #121: > > @barndawgie thanks for sharing. I keep meaning to look in to this. Everyone really should consider moving to this, this integration is only going to continue to break over time, we aren't using an open API and we all know how cloud products are... this is one of the two devices I have that still use a cloud connection.

barndawgie, 2023-10-28:
> > @barndawgie thanks for sharing. I keep meaning to look in to this. Everyone really should consider moving to this, this integration is only going to continue to break over time, we aren't using an open API and we all know how cloud products are... this is one of the two devices I have that still use a cloud connection.
> > Also, last I heard electric water heaters weren't working? Assume there has been movement there based on your comment?
> 
> Yes! That is a recently added feature, compliments of @tronikos.
> 
> EDIT: I should add that I only added this to my setup in the last 3-6 months. It is probably one of the easiest ESPHome projects to get into because you're basically buying spec hardware and just attaching it to an existing jack.

has anyone created a youtube video of the process to create the device and program for simplicity?

_Originally posted by @fireheadman in https://github.com/home-assistant/core/issues/102800#issuecomment-1783847529_

Ylianst, 2023-11-11:
I recorded a video tonight as I installed this project on my Rheem v4 Heat pump water heater. I have to edit it and will port here in the next few days.

Ylianst, 2023-11-11:
Youtube is still processing the video as I type this, but here is the link: [https://www.youtube.com/watch?v=4IVNuJEpytA](https://www.youtube.com/watch?v=4IVNuJEpytA).

I have a Rheem Gen 4 heat pump water heater and connected it using a ATOM device from R5Stack. Everything worked perfectly. I also have a Moen Flo and a hot water re-circulation pump attached to a Zigbee plug.

fireheadman, 2023-11-13:
awesome work!
now just need to get a good parts list, unsure if I can source these from AMZ

maybe something like these?
[https://www.amazon.com/M5Stack-AtomS3-Dev-0-85-inch-Screen/dp/B0C7KYYF5S](url)
[https://www.amazon.com/M5StickC-Plus-ESP32-PICO-Mini-Development/dp/B08VGST8LJ](url)

fireheadman, 2023-11-13:
Actually... found it here instead so ordered the kit.
Hopefully it doesn't take too long to arrive

[https://thepihut.com/products/atom-rs485-kit?variant=40639072600259](url)

[image]

Ylianst, 2023-11-14:
There are rumors that these are not manufactured anymore, so if you find one, order it right away. Going forward, we many need to find alternative hardware.

barndawgie, 2023-11-16:
@Ylianst - Thanks so much for making that video! The production values are surprisingly good :-)

I don't suppose you'd be open to editing one using [the web installer](https://esphome-econet.github.io/install/)? I think that would be much quicker/easier for uses, and also result in a much shorter video that we could post in the README.

Ylianst, 2023-11-17:
@barndawgie Oh, how did I not know about this. Sure, I can give that a try!

barndawgie, 2023-11-17:
> @barndawgie Oh, how did I not know about this. Sure, I can give that a try!

It is newly enabled for this project in the last week.

Ylianst, 2023-11-23:
I just added a [YouTube short version of my original video](https://youtube.com/shorts/J4Zf1Vtf2CA). It's a 58 second version.

barndawgie, 2023-11-24:
That's a nice teaser trailer for the integration! :-)

Would still be great, if you can, to get one where you install via the website as that would bypass steps like installing the ESPHome dashboard, etc...

Ylianst, 2023-11-24:
Yes. I plan on recording the video this evening. By the way, I love this project. Here is what one week of temperature data looks like on my system.

[image]

Ylianst, 2023-11-25:
New YouTube video: [https://www.youtube.com/watch?v=5u0_TsBOTEI](https://www.youtube.com/watch?v=5u0_TsBOTEI)
New short video: [https://www.youtube.com/shorts/K75njuvRIks](https://www.youtube.com/shorts/K75njuvRIks)

barndawgie, 2023-11-25:
Awesome! Thanks so much!

closed 2023-11-25 by barndawgie

fireheadman, 2023-12-06:
Misplaced this link, finally found this again!
Wanted to share photos from this install.

I used the web installer also, took 3 min to completed, very nice!!!

https://github.com/esphome-econet/esphome-econet/discussions/118#discussioncomment-7769751

[image]