esphome-econet/esphome-econet #130: Generic Setup Help: ESP32 + RS485 to TTL (5V Board /w MAX13487 Chip)

PrinceAli0, 2023-11-02:
First off thanks again for this wonderful project.  I've had spotty performance with the OOB ECONET app/integration on my Rheem Hybrid Water Heater.  I've actually ordered a M5Stack A131 and M5Stack C008 Atom Lite but it's going to be a while before I get it.  I happen to have spare parts from other projects and based on our statement that we can use many other ESP32/ESP8266 + an RS485 converter, so off I went.  For reference, these are what I have:

- ESP32 - [ESP Module](https://www.amazon.com/ACEIRMC-Development-Dual-Mode-Microcontroller-Integrated/dp/B09YY6W3KD?th=1)
- RS485 - [RS485 Module](https://www.amazon.com/DSD-TECH-SH-U12-MAX13487-Raspberry/dp/B07B667STP)

I had used ESPHome on my HomeAssistant instance early on when I built some sensors and re-added the add-on.  Ran though the process: Build a **New Device > Name > ESP32 > Copied Encryption Key**.  Once the device config was created, I edited the YAML file and adding in the code you've got:

```esphome:
  name: sh-water-heater
  friendly_name: SH-Water-Heater

substitutions:
  tx_pin: GPIO1
  rx_pin: GPIO3
  platform: esp32
  board: esp32dev

packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    # Change this file to the correct one for your device
    file: econet_heatpump_water_heater.yaml
    # file: econet_hvac.yaml
    # file: econet_tankless_water_heater.yaml
    # file: econet_electric_tank_water_heater.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXYYYZZZ111222333444555"
```

I then range the **Install > Manual Download > Modern Format >** download the file and **Installed the BIN file** via https://web.esphome.io/.   Once completed, I wired up the cables as follows:
### GPIO1/GPIO3 Setup
[image]

In my case, I used an RJ12 for the serial connection to the console/serial port on the Rheem Water Heater.  Once everything was wired up, I connect the device via a USB-C power and waited for it to successfully show up on my network (UniFi WiFi).  While it comes online and I'm able to add the ESPHome integration and add my custom device (via IP) to Home Assistant.  However, I can see the module connecting into HomeAssistant but it appears that nothing is working (i.e. _changing the mode doesn't work, changing the temperature doesn't work_).  

In the image above, you can see I used the UART0 ports on the ESP32.  Should I be using UART1 (GPIO9, GPIO10) or UART2 (GPIO16, GPIO17) for the wiring (and subsequent updates to the YAML tx_pin and rx_pin).  Trying to see if I can get this working using the parts I have (curiousness and potentially helping others who do not get an M5Stack ATOM RS485 K045 Kit).

Any help/direction will be helpful and I'm open to sharing my documentation/guide to help others (not sure if you're looking for a more detailed guide....a novice but I'm willing to help something that I'm interested in).

tronikos, 2023-11-02:
See https://github.com/esphome-econet/esphome-econet/wiki/Schematics the section about the flow control pin. Let us know if you encounter any issues since I submitted this today and it's untested.

PrinceAli0, 2023-11-02:
@tronikos, I saw that post (well earlier revisions of it and wondered if that might be my issue).  I will absolutely connect my RS485 to 3.3V vs 5V as you've instructed and will test.  However, I don't believe my RS485 module has DE/RE pins/solder points.

tronikos, 2023-11-02:
Sorry I was on mobile and your RS485 module looked like the one with flow control pin. Your wiring (you have a typo in your diagram on the ground wire) and your yaml seems fine. Hopefully switching to 3.3V as recommended in the wiki will work. If not I'm out of ideas. You might want to join https://discord.gg/cRpxtjkQQ3 and ask the community there.

elmoret, 2023-11-02:
@PrinceAli0 Are you sure you have RX/TX correct? Happy to double check a photo of the actual wiring.

PrinceAli0, 2023-11-03:
Apologies for the delay in responding.  

@elmoret, see the following images representing my physical setup.  Hopefully this covers what you're looking for to validate if I got everything correct:

**Full Setup**
[image]

**ESP32 -> Wire**
[image]

**Wire -> RS485**
[image]

**RS485 > RJ12 Cable**
[image]

@tronikos, thanks for catching my typo in the diagram.  I've updated it for accuracy and reference for anyone else trying to get this going with an ESP32 + RS485 module.

elmoret, 2023-11-03:
Pretty sure TX goes to RX and vice versa, though I can't say for sure without a schematic of that RS485 module. Won't hurt anything to try.

The MAX13487 has a minimum voltage of 4.75v, not sure how it'd behave at 3.3v.

Try swapping TX/RX and report back?

tronikos, 2023-11-03:
The note "Please make sure to power the RS485 module with 3.3V because it affects the TTL (transistor-transistor logic) voltage between RS485 module and ESP." in https://github.com/esphome-econet/esphome-econet/wiki/Schematics was copied from https://github.com/syssi/esphome-solax-x1-mini#schematics
No idea if it's required.
@elmoret while we have you here, since you seem knowledgable of electronics, could you please review https://github.com/esphome-econet/esphome-econet/wiki/Schematics ?

tronikos, 2023-11-03:
FWIW, I have the following in my yaml:
```yaml
substitutions:
  tx_pin: GPIO4
  rx_pin: GPIO5
  platform: esp8266
  board: d1_mini
```

I used GPIO4 (D2) as TX pin and GPIO5 (D1) as RX pin instead of the regular TX/RX because of 
https://github.com/syssi/esphome-solax-x1-mini/blob/58df8586d2260cbdf50046af630774d8e5ad9b41/esp8266-example.yaml#L5-L6

No idea why the config in that package chose GPIO4 and GPIO5, when the schematics in the same package recommend using the TX/RX pins. I personally avoided the TX/RX because they are both high at boot.
@PrinceAli0 maybe it's worth trying pins other than TX/RX?

I have wired the GPIO4 to the TX pin at HW-0519 and the GPIO5 to the RX pin at HW-0519, see photo.
[image]

@elmoret why did you suggest swapping TX/RX for the MAX13487?

elmoret, 2023-11-03:
My suggestion wasn't MAX13487 specific, its just that the manufacturers of these modules do "interesting" things with the labels sometimes. But proper convention is that TX connects to RX, and vice versa.

https://microcontrollerslab.com/uart-communication-working-applications/

It makes sense, right? One side transmits to the other side's receive...

PrinceAli0, 2023-11-03:
So swapping TX/RX (and cycling between 5V and 3.3V) didn't bring this online.

@tronikos , thanks for the feedback.  I will try some of the other GPIO ports suggested for my ESP32 module for UART from the table:

[image]

Unless you suggest me using other pins not tagged/suggested for TX/RX operations.

One last thing....maybe I'm getting in my own head but I'm trying to figure out if I have a **Rheem EcoNet Electric Tank Water Heater** or a **Rheem EconNet HeatPump Water Heater**.  This of course is something I have to choose within the YAML file.  My model is XE50T10HD50U1 and the documentation (https://images.homedepot-static.com/catalog/pdfImages/fe/fea1a916-5918-44dc-ada3-f5ea733203d1.pdf)  suggests the following details that is confusing me (highlighted in yellow):

[image]

[image]

Regardless, going to try what @tronikos has suggested with the GPIO pins and report back shortly.

elmoret, 2023-11-03:
That's a heat pump waterheater. You'll know for sure if it blows out air when running :)

PrinceAli0, 2023-11-03:
@elmoret, thanks for that simple explanation....10 mins of bouncing around the Rheem website couldn't give me such a quick answer.  So I have this device set correctly (_econet_heatpump_water_heater.yaml_) in my YAML.  I do enjoy that fan blowing in the garage when the kids are taking a bath and I'm getting cooled off while tinkering out there :)

[image]

Now...some great news!  It all works.  I didn't realize both my test prior (going to 3.3V & reversing TX/RX) failed due to my UniFi UDM Pro ignoring the Static IP assignment to the ESP32 device when I reconnected (a simple **_ping_** test pointed it out).  Now that I've got it behaving as it should, we're reporting everything successfully:

For reference, the following is the working configuration for anyone chasing this with similar parts/config.  

**Design**
[image]

**Physical Wiring**
[image]

I want to extend a big thanks to both @elmoret and @tronikos for helping me out and making this a success.  I can check yet another Smart Device off the cloud connected list :)

PrinceAli0, 2023-11-03:
@tronikos, I also have an M5Stack ATOM RS485 K045 Kit on order (who knows when it will get here).  I will swap out my ESP32 +  RS485 to TTL (5V Board /w MAX13487 Chip) for that when it arrives.

Do you have any interest in me sharing/documenting the logical diagram and physical images with annotations for that build out.  I'm going to do it for my own purposes/tracking so I can refer back to my effort in future (or my kids can if they're interested). 

Just let me know and I'll post it to this thread and/or send it your way. 

Thanks again for all those that made this wonderful project :)

tronikos, 2023-11-04:
Great to hear you fixed it. I'm closing this. I added the following at https://github.com/esphome-econet/esphome-econet/wiki/Schematics

Note: for some RS485-TTL modules such as MAX13487 you need to connect the RX pin of the ESP board to the TX pin of the RS485-TTL module, and the TX pin to the RX pin, see wiring with photos [here](https://github.com/esphome-econet/esphome-econet/issues/130).

@PrinceAli0 you welcome to send a pull request to update the readme or the wiki with whatever information you think would help new users.

closed 2023-11-04 by tronikos

wjschmitt, 2023-12-16:
I had to connect to RX2/TX2 instead of RX0/TX0 because I was getting a serial port conflict with the logger.