esphome-econet/esphome-econet #164: Add Sesnsor for Rheem Microcontroller Time

barndawgie, 2023-11-16:
Per discussion in Discord, sounds like it's possible to determine the Rheem Microcontrollers current time: https://discord.com/channels/1148015790038188073/1151284280408948868/1174744288291201094

I think this would be useful to add a Diagnostic Sensor.

barndawgie, 2023-11-21:
@tronikos - Is this possible with existing code? Or we'd need to expose some additional data from the C layer?

tronikos, 2023-11-21:
It depends on how it's exposed. Is it available?

barndawgie, 2023-11-21:
There is no string for it that I'm aware of, but @Daniel-dev22 said he thought it came back on other calls?

tronikos, 2023-11-21:
There is occasional writes with time, it seems to make sure communicated units are in sync, see https://github.com/esphome-econet/esphome-econet/blob/72d640ad6998cd6d116d10beb045f614f85904ca/components/econet/econet.cpp#L261-L262
I don't think it makes sense to expose that given how infrequent they are. And they are only present if there is a thermostat. It would be better to periodically issue such a write command from our component.

closed 2023-11-21 by barndawgie