esphome-econet/esphome-econet #168: Setting Econet Update Interval Doesn't Change Interval

stockmopar, 2023-11-19:
Having a faster update time is beneficial, especially for the tankless water heater data. When trying to set a faster value using the substitution econet_update_interval it does not appear to change the request rate.

tronikos, 2023-11-20:
At what value did you set it? I just tried setting it to 10s and it worked as expected. Note that if the value is unchanged it won't publish it to Home Assistant. If you set logger_level to VERBOSE or DEBUG do you see the requests made when they are supposed to?
Why does it need to be more frequent than 30s for tankless water heater? Should we change the default for all or just for tankless?

barndawgie, 2023-11-20:
Since tankless units heat water on demand it makes sense that having more realtime data for them (compared to other units) would be useful.

stockmopar, 2023-11-20:
Flow rate and instant btus (which depends on flow, temp in, temp out) are the big ones that are beneficial to have at the faster rates. I'm trialing adding gas flow meters that will show up in the energy dashboard.

tronikos, 2023-11-20:
> Since tankless units heat water on demand it makes sense that having more realtime data for them (compared to other units) would be useful.

Ah yes that makes sense.

barndawgie, 2023-11-21:
So is this not working for you, @stockmopar? Where are you seeing slow updates, in the ESP logs or HA?

stockmopar, 2023-11-22:
It's working! Looks like I was using the substitution wrong. Closing!

closed 2023-11-22 by barndawgie

tronikos, 2023-11-22:
I noticed the template sensor was missing update_interval so it was using the default 60s. That's fixed in https://github.com/esphome-econet/esphome-econet/pull/182
https://github.com/esphome-econet/esphome-econet/pull/183 reduces the default interval to 5s