esphome-econet/esphome-econet #172: Power entity missing for electric water heater

vangorra, 2023-11-20:
This component works great. I see there is documentation for reading power consumption (https://github.com/esphome-econet/esphome-econet/wiki/Monitoring-Power-Consumption-in-kWh). However, my installation does not have a power entity. How does one go about getting the power entity for this water heater?

Type: Electric Tank Water Heater
Model: XE50T12CS55U1

Build config:
```
---
substitutions:
  tx_pin: GPIO19
  rx_pin: GPIO22
  platform: esp32
  board: esp32dev

packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    file: econet_electric_tank_water_heater.yaml

dashboard_import:
  package_import_url: github://esphome-econet/esphome-econet/build-yaml/${name}-${platform}.yaml@${github_ref}
  import_full_config: false

# Uncomment the below to use Wi-Fi settings from your secrets.yaml file
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
  encryption:
    key: !secret encryption_key

```

Home assistant device:
[image]
[image]
[image]

barndawgie, 2023-11-21:
@vangorra - @Daniel-dev22's question is about the official EcoNet app, to be clear. We're curious if you see power consumption in there as it would imply that there is some API/value for us to query against.

vangorra, 2023-11-21:
I figured as much. I think the app used to show power consumption but I'm not sure.
To be sure, I'm trying to get the app and unit to connect again. I stopped using all this shortly after install because it's all so terrible and I'm reminded again how awful their systems are.

vangorra, 2023-11-21:
Finally managed to get this stupid "smart" thing to connect. It does show power usage. 

 [image]

barndawgie, 2023-11-21:
You can also try copying the POWRWATT sensor from https://github.com/esphome-econet/esphome-econet/blob/main/econet_heatpump_water_heater.yaml to try a quick test.

vangorra, 2023-11-22:
That worked. Using the config below, I was able to add a power reading for my electric water heater.

Rheem Performance Platinum Gladiator
Model  XE50T12CS55U1

```
---
substitutions:
  tx_pin: GPIO19
  rx_pin: GPIO22
  platform: esp32
  board: esp32dev
  sensor_power_filters_lambda: return x;

packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    file: econet_electric_tank_water_heater.yaml

dashboard_import:
  package_import_url: github://esphome-econet/esphome-econet/build-yaml/${name}-${platform}.yaml@${github_ref}
  import_full_config: false

# Uncomment the below to use Wi-Fi settings from your secrets.yaml file
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
  encryption:
    key: !secret encryption_key

sensor:
  - platform: econet
    name: "Power"
    id: power
    sensor_datapoint: POWRWATT
    unit_of_measurement: "W"
    accuracy_decimals: 3
    device_class: "power"
    state_class: "measurement"
    filters:
      - lambda: ${sensor_power_filters_lambda}
```

barndawgie, 2023-11-22:
Just submitted a PR to add this to our configurations.

@vangorra - If you're up to trying some of the other sensors from that file, it's definitely possible that some of them would also apply to an ETWH (e.g. The Model and Serial Number sensors, some of the temperature diagnostic sensors).

closed 2023-11-22 by barndawgie

tronikos, 2023-11-22:
@vangorra Do VACA_NET and VACATIME that are in the HPWH config work for your water heater? Are you missing any presets? The current config for electric water heater was based on an old unit that only had:

```
    custom_presets:
      0: "Energy Saver"
      1: "Performance"
```

vangorra, 2023-11-22:
Looks like vacation mode is working when I copied it over from the heat pump config. I created a PR with the changes.

The existing 2 presets do work but I'm not sure if it is a complete list. I tried to look in the app but it of course doesn't work again and re-adding the unit from scratch also isn't working anymore. Who would have guessed that a water tank manufacturer is terrible at making connected devices and apps?!

barndawgie, 2023-11-22:
I would be surprised if there are other modes - all of the other modes on the HPWH are about how aggressively the heater switches from Heat Pump to Heating Coils.

tronikos, 2023-11-22:
From https://github.com/esphome-econet/esphome-econet/wiki/Objects you can try adding the following in your yaml

```yaml
text_sensor:
  - platform: econet
    name: "WHTRCNFG text"
    sensor_datapoint: WHTRCNFG

number:
  - platform: econet
    name: "WHTRCNFG number"
    number_datapoint: WHTRCNFG
    min_value: 0
    max_value: 100
    step: 1
    mode: box
```

and try different values for the number. If you find any other working, just add the `number: "text"` in the `custom_presets` map.

vangorra, 2023-11-22:
Thanks for the config. I tried it out and it looks like there aren't any additional presets.
```
[21:22:52][W][econet.climate:096]: In custom_presets of your yaml add: 2: ""
[21:22:52][W][econet.climate:096]: In custom_presets of your yaml add: 3: ""
[21:22:53][W][econet.climate:096]: In custom_presets of your yaml add: 4: ""
[21:22:54][W][econet.climate:096]: In custom_presets of your yaml add: 5: ""
[21:22:56][W][econet.climate:096]: In custom_presets of your yaml add: 6: ""
[21:22:57][W][econet.climate:096]: In custom_presets of your yaml add: 7: ""
[21:22:58][W][econet.climate:096]: In custom_presets of your yaml add: 8: ""
[21:22:58][W][econet.climate:096]: In custom_presets of your yaml add: 9: ""
```

tronikos, 2023-11-22:
Thanks for checking. As barndawgie said, it makes sense there aren't any more supported presets.