esphome-econet/esphome-econet #188: Objects of Length 1 are not properly decoded

stockmopar, 2023-11-24:
In the follow snippet of log, PRES is not decoded:

`[07:45:59][I][econet:165]: <<< 80.00.03.80.00.80.00.03.40.00.B6.00.00.1E.02.01.00.00.53.50.54.00.00.00.00.00.00.00.52.45.4C.48.37.30.30.35.00.00.50.52.45.53.5F.53.55.43.00.00.50.52.45.53.00.4C.49.51.00.00.54.45.4D.50.5F.4F.53.54.00.00.54.45.4D.50.5F.4F.4C.54.00.00.54.45.4D.50.5F.45.56.50.00.00.54.45.4D.50.5F.53.53.54.00.00.53.55.42.5F.43.4F.4F.4C.00.00.54.45.4D.50.5F.4C.41.54.00.00.54.45.4D.50.5F.52.41.54.00.00.50.52.45.53.5F.53.55.41.00.00.54.45.4D.50.5F.53.53.41.00.00.54.45.4D.50.5F.4F.53.41.00.00.49.4E.44.
[07:45:59][I][econet:166]:   Dst Adr : 0x380
[07:45:59][I][econet:167]:   Src Adr : 0x340
[07:45:59][I][econet:168]:   Command : 30
[07:45:59][I][econet:169]:   Data    : 02.01.00.00.53.50.54.00.00.00.00.00.00.00.52.45.4C.48.37.30.30.35.00.00.50.52.45.53.5F.53.55.43.00.00.50.52.45.53.00.4C.49.51.00.00.54.45.4D.50.5F.4F.53.54.00.00.54.45.4D.50.5F.4F.4C.54.00.00.54.45.4D.50.5F.45.56.50.00.00.54.45.4D.50.5F.53.53.54.00.00.53.55.42.5F.43.4F.4F.4C.00.00.54.45.4D.50.5F.4C.41.54.00.00.54.45.4D.50.5F.52.41.54.00.00.50.52.45.53.5F.53.55.41.00.00.54.45.4D.50.5F.53.53.41.00.00.54.45.4D.50.5F.4F.53.41.00.00.49.4E.44.4F.4F.52.4B.57.00.00.49.53.43.53.5
[07:45:59][I][econet:184]:   Type    : 2
[07:45:59][I][econet:185]:   PropType: 1
[07:45:59][I][econet:199]:   SPT
[07:45:59][I][econet:199]:   RELH7005
[07:45:59][I][econet:199]:   PRES_SUC
[07:45:59][I][econet:199]:   PRES
[07:45:59][I][econet:199]:   LIQ
[07:45:59][I][econet:199]:   TEMP_OST
[07:45:59][I][econet:199]:   TEMP_OLT
[07:45:59][I][econet:199]:   TEMP_EVP
[07:45:59][I][econet:199]:   TEMP_SST
[07:45:59][I][econet:199]:   SUB_COOL
[07:45:59][I][econet:199]:   TEMP_LAT
[07:45:59][I][econet:199]:   TEMP_RAT
[07:45:59][I][econet:199]:   PRES_SUA
[07:45:59][I][econet:199]:   TEMP_SSA
[07:45:59][I][econet:199]:   TEMP_OSA
[07:45:59][I][econet:199]:   INDOORKW
[07:45:59][I][econet:199]:   ISCSPEED
[07:45:59][I][econet:199]:   FURNSTMP
[07:45:59][I][econet:199]:   FURNRTMP
[07:45:59][W][component:214]: Component econet took a long time for an operation (0.18 s).
[07:45:59][W][component:215]: Components should block for at most 20-30ms.
[07:45:59][I][econet:342]: Read 380. ms=222590
[07:45:59][I][econet:165]: <<< 80.00.03.40.00.80.00.03.80.00.8A.00.00.06.07.80.00.00.42.84.A1.85.07.80.00.00.42.1C.E7.A3.07.80.00.00.00.00.00.00.01.15.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.01.00.01.01.07.80.00.00.00.00.00.01.07.80.00.00.43.8D.0B.93.07.80.00.00.00.00.00.00.07.80.00.00.C2.20.00.00.07.80.00.00.42.8A.CC.CD.AF.A3 (154)
[07:45:59][I][econet:166]:   Dst Adr : 0x340
[07:45:59][I][econet:167]:   Src Adr : 0x380
[07:45:59][I][econet:168]:   Command : 6
[07:45:59][I][econet:169]:   Data    : 07.80.00.00.42.84.A1.85.07.80.00.00.42.1C.E7.A3.07.80.00.00.00.00.00.00.01.15.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.00.00.00.00.07.80.00.00.01.00.01.01.07.80.00.00.00.00.00.01.07.80.00.00.43.8D.0B.93.07.80.00.00.00.00.00.00.07.80.00.00.C2.20.00.00.07.80.00.00.42.8A.CC.CD (138)
[07:45:59][I][econet:281]:   SPT : 66.315468
[07:45:59][I][econet:281]:   RELH7005 : 39.226208
[07:45:59][I][econet:281]:   PRES_SUC : 0.000000
[07:45:59][I][econet:281]:   LIQ : 0.000000
[07:45:59][I][econet:281]:   TEMP_OST : 0.000000
[07:45:59][I][econet:281]:   TEMP_OLT : 0.000000
[07:45:59][I][econet:281]:   TEMP_EVP : 0.000000
[07:45:59][I][econet:281]:   TEMP_SST : 0.000000
[07:45:59][I][econet:281]:   SUB_COOL : 0.000000
[07:45:59][I][econet:281]:   TEMP_LAT : 0.000000
[07:45:59][I][econet:281]:   TEMP_RAT : 0.000000
[07:45:59][I][econet:281]:   PRES_SUA : 0.000000
[07:45:59][I][econet:281]:   TEMP_SSA : 0.000000
[07:45:59][I][econet:281]:   TEMP_OSA : 282.090424
[07:45:59][I][econet:281]:   INDOORKW : 0.000000
[07:45:59][I][econet:281]:   ISCSPEED : -40.000000
[07:46:00][I][econet:281]:   FURNSTMP : 69.400002`

stockmopar, 2023-11-24:
In finding that this PRES[00]LIQ appears to be an invalid object string. I'm not sure what the meaning of an object with just a value of 0x15 mean. It could be a value... or... it could be an indicator of that string is not supported.

stockmopar, 2023-11-24:
In throwing in some random unsupported strings, it appears this is the response for strings that are not supported!

barndawgie, 2023-11-24:
Would be great to build
Some support for this, actually, as I've been using the absence of a return value to mean this but a definitive answer would be better.

closed 2023-11-25 by stockmopar