esphome-econet/esphome-econet #192: Setting an Enum from a Select Can Causes the Text Value to be Blank (internally)

stockmopar, 2023-11-25:
When you set a value from a select it successfully writes to the device. But internally it updates the datapoint with an empty enumerated text. This causes subsequent refreshes of data to appear to not have any changes in value with the text continuing to be blank.

This can be reproduced by creating a Select and a Text Sensor of the same type. If you change the value with the Select, the Text Sensor will go blank.

stockmopar, 2023-11-25:
This can be worked around in various cases. I ran into this when trying to setup the Fan Mode switching between the scheduled variant (STAT_FAN) and non-scheduled (STATNFAN) based on the follow schedule object (SCHEDULE). 

I had a text sensor for SCHEDULE setup, a select sensor for SCHEDULE, and SCHEDULE tied to the climate sensor.

I initially tried retaining the value of SCHEDULE in the climate sensor by its text value, but since when you change the value with the select sensor it sends out an update to the other components with send_datapoint_() and updates the send_datapoint_[datapoint_id] with the new value it causes this issue.

tronikos, 2023-11-25:
Can you try changing https://github.com/esphome-econet/esphome-econet/blob/15a64f2862c2a9573f80fdb886cd992e1bbf785d/components/econet/econet.h#L53 from `return lhs.value_enum == rhs.value_enum;` to `return lhs.value_enum == rhs.value_enum && lhs.value_string == rhs.value_string;`?

stockmopar, 2023-11-28:
This change does fix it when the datapoint is refreshed again. Before that, a datapoint gets pushed through and potentially displayed as blank.

I'd be fine with this as the solution, if this ends up causing problems down the road for a use case a new issue can be opened for it.

PR #219 (2023-11-29): Fix #192
Fix #192
commit adf609b3 (2023-11-29): Fix #192 (#219)
closed 2023-11-29 by tronikos