esphome-econet/esphome-econet #200: Add humidity and aux heat support to the HVAC climate entity

tronikos, 2023-11-26:
Blocked on https://github.com/home-assistant/core/pull/103807

stockmopar, 2023-12-18:
FYI. @tronikos  @Daniel-dev22 

 It looks like aux heat support was remove from the HVAC climate entity. But humidity support was added (and is now merged both in Home Assistant and ESPHome. I would assume when we do add support for this, it would mean we are dependent on users having newer versions of HA and ESPHome.

barndawgie, 2023-12-21:
Looks like Humidity support was added to ESPHome here: https://github.com/esphome/esphome/releases/tag/2023.12.0

So we should be able to add this now.

stockmopar, 2023-12-26:
Removed aux heat as it appears that it appears that part is deprecated by ha and not planned now for esphome.

closed 2023-12-26 by stockmopar