esphome-econet/esphome-econet #209: PoESP32 econet-base.yaml

justbruce, 2023-11-28:
Changes made to create a working Power over Ethernet esphome-econet node.

https://github.com/esphome-econet/esphome-econet/compare/main...justbruce:esphome-econet:patch-1

barndawgie, 2023-11-29:
I _think_ what you'd actually want to do is create versions of each YAML under the [build-yaml folder](https://github.com/esphome-econet/esphome-econet/tree/main/build-yaml) for these, e.g. `econet-etwh-poesp32.yaml`, `econet-hpwh-poesp32.yaml`, etc...

The tricky thing is you can't just remove stuff from econet_base since that is where all of the devices currently pick them up. It might be possible to disable them (e.g. `wifi:` and `wifi-signal:`) with an !extends keywords, but I'm not exactly sure.

barndawgie, 2023-11-29:
You might want to join our Discord to discuss this, @justbruce: https://discord.gg/zCRquauF

justbruce, 2023-11-29:
> You might want to join our Discord to discuss this, @justbruce: https://discord.gg/zCRquauF

I fairly loathe Discord since so much of what should be public, and useful to other folks, ends up locked behind Discords walls.

  I will, however, give it some thought.  You folks seem likeable enough.

justbruce, 2023-11-29:
> I _think_ what you'd actually want to do is create versions of each YAML under the [build-yaml folder](https://github.com/esphome-econet/esphome-econet/tree/main/build-yaml) for these, e.g. `econet-etwh-poesp32.yaml`, `econet-hpwh-poesp32.yaml`, etc...
> 
> The tricky thing is you can't just remove stuff from econet_base since that is where all of the devices currently pick them up. The ability to remove them from the package doesn't exist today, but will be added shortly, looks like: [esphome/esphome#5821](https://github.com/esphome/esphome/pull/5821)

I'll be happy to write up the config file when this "!remove" directive makes it into esphome.

  For now, I will continue to locally compile.

barndawgie, 2023-11-29:
Okay - let's keep this bug open and we can come back to it once it's implemented and at least provide sample config if not a "supported" build.

barndawgie, 2023-11-29:
Was setting the `board` important to get this working? https://esphome.io/components/esp32 says "This only affects pin aliases, flash size and some internal settings, if unsure choose a generic board from Espressif such as esp32dev" so I was curious how much it matters.

justbruce, 2023-11-29:
> Was setting the `board` important to get this working? https://esphome.io/components/esp32 says "This only affects pin aliases, flash size and some internal settings, if unsure choose a generic board from Espressif such as esp32dev" so I was curious how much it matters.

It seemed to be.  Flashing would fail if I left it as 'esp32dev'

justbruce, 2023-12-07:
Looks like !remove made it into esphome yesterday.  I'll see if I can't wrap my brain around how it works

justbruce, 2023-12-10:
> Looks like **!remove** made it into esphome yesterday. I'll see if I can't wrap my brain around how it works

I have the build-yaml files made up for whenever the !remove directive makes it into the release version of esphome

barndawgie, 2023-12-16:
This is the same ESP32 chip as other units, right? So you have wi-fi but are choosing to use Ethernet instead? I'm not sure it makes sense for us to include this in what we build unless this is a widely used scenario.

Probably makes more sense to just include it as an example configuration somewhere in the repo or wikis.

justbruce, 2023-12-16:
> This is the same ESP32 chip as other units, right? So you have wi-fi but are choosing to use Ethernet instead? I'm not sure it makes sense for us to include this in what we build unless this is a [...]

I'm not actually sure it is the same chip - according to the supplier, the WiFi is disabled and unusable.  I opened my unit and there is no antenna on the PCB, and no antenna connection.

I use POE because [among other reasons] I mounted the unit amongst the maze of metal ducting and enclosures, and reliably getting signal into the nearly complete faraday cage seemed.... dubious.

  Either way, I won't be super upset.  I prepped the files, but it only took 10 minutes, so no big loss

barndawgie, 2023-12-16:
At a minimum we can include one in the wiki or some directory of examples configs.

justbruce, 2023-12-16:
I like example configs directory.  That way anyone who needs it can pull the file but it doesn't get continually built

barndawgie, 2023-12-17:
> I like example configs directory. That way anyone who needs it can pull the file but it doesn't get continually built

The main issue is I don't want to include it in the Web Tools if it's yet another option to have to check on that page (which it would be if it still shows as an ESP32 device).

I think there are other sample configs folks have (I know have one) so if you create a directory or wiki for them there will be more content :-)

justbruce, 2023-12-17:
As in....start a separate project?  I don't have write access to this one so I assume that is what you mean....

barndawgie, 2023-12-17:
Submit a PR!

closed 2023-12-17 by justbruce

barndawgie, 2023-12-17:
@justbruce can you update the wiki? Or don't have permissions?

barndawgie, 2023-12-17:
Added this wiki page with reference code:
https://github.com/esphome-econet/esphome-econet/wiki/Configuration-Examples

justbruce, 2023-12-18:
Thanks.

  I don't know if I will have time to maintain the wiki, but I do appreciate the work you put into it.

justbruce, 2024-03-01:
> In regards to the example (and forgive me for my lack of understanding so far...), I have the Atom Lite and I want to turn off the wireless and use only ethernet (and POE I suppose would be nice but I don't have a switch yet).
-snip-
> On the grove connector: G26 G32 ^ these will be my mdc_pin: GPIO32 mdio_pin: GPIO26 ?? Or visa versa?

Without knowing which ETH module [and thus, which chip] you are trying to use, I can't say.  Which pins will you be using to interface with the RS485 module?

barndawgie, 2024-03-04:
> In regards to the example (and forgive me for my lack of understanding so far...), I have the Atom Lite and I want to turn off the wireless and use only ethernet (and POE I suppose would be nice but I don't have a switch yet).  
> 
> 
> 
> [image]
> 
> 
> 
> On the grove connector:
> 
> G26
> 
> G32
> 
> ^ these will be my 
> 
> mdc_pin: GPIO32
> 
> mdio_pin: GPIO26
> 
> ??  Or visa versa?
> 
> 

Not sure the exact specifics but we have an example on the wiki that worked for one user: https://github.com/esphome-econet/esphome-econet/wiki/Configuration-Examples#use-ethernetpoe-instead-of-wi-fi

iamburgerking, 2024-03-04:
I feel like the image I posted isn't being shown??  I'll try to illustrate in text.

The RS485 module (M5Stack C008 ATOM Lite) has the options of 4 connections.  I get the two obvious ones power and ground, but the other 2, I'd like some clarity as the example in the wiki "sorta" points me in the right direction, but I'm dim still.

== GRD == - == 5V== - ==G26== - ==G32==

So from the example:
ethernet:
  type: IP101
  mdc_pin: GPIO23    <---- is this G26?
  mdio_pin: GPIO18    <----is this G32?
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO5

Thanks for the patience!

barndawgie, 2024-03-04:
Rather than posting in this closed issue, you might have more luck if you came over to our Discord: https://discord.gg/SNd3YsSKtG