esphome-econet/esphome-econet #211: Modify "Hardware Installation" section of Wiki for wiring clarity

justbruce, 2023-11-28:
Add/Substitute the highlighted text

>*The ESPHome device communicates to the the Econet enabled device thru the RJ11/12 jack ONLY.  Do not attempt to wire into serial terminals on control boards or thermostats as this will cause instability and communications failures between devices*

>Cut the connector off one one end of the RJ11/12 cable, then strip and connect the wires to the RS485 device's screw terminals. Pin 3 to B, Pin 4 to A, and Pin 5 to GND. A and B are for data communication, and G is ground. 12V is left empty.

>NOTE: Wire colors can vary from cable to cable. Ensure you are matching the pins as shown in the diagram below regardless of cable color, i.e. Pin 3 to B, Pin 4 to A, Pin 5 to G.

barndawgie, 2023-12-22:
Why do you think this update is necessary? Is there some place in the docs that it suggests wiring directly into other connections?

justbruce, 2023-12-22:
I'll only say that it was NOT clear when I did my install that you shouldn't do this.  In matter of fact, I did wire directly into those terminals, as the RJ-11 jack was hidden by the cable bundle from the initial installer.   This is how I know that the system will react poorly to such a configuration

  In my mind, anything that can damage the system should be called out multiple times, in multiple ways.

  Just my $.02.

barndawgie, 2024-01-29:
@bitfliq made some recent updates so I'm considering this complete.

closed 2024-01-29 by barndawgie