esphome-econet/esphome-econet #228: Add Energy Consumption Integrations for Electric Devices

barndawgie, 2023-11-30:
https://github.com/esphome-econet/esphome-econet/pull/224/ adds total gas consumption for furnaces. Would be great to provide a similar sensor for other units - I currently create one in my HA config but providing it from ESPHome-econet would be cleaner.

justbruce, 2024-04-03:
For HVAC, try ELEC_AVG and ELEC_DAY.  

I don't have the requisite equipment [my secondary heat is gas/dual fuel] so I don't know if these are the appropriate points

justbruce, 2025-03-11:
> For HVAC, try ELEC_AVG and ELEC_DAY.
> 
> I don't have the requisite equipment [my secondary heat is gas/dual fuel] so I don't know if these are the appropriate points

There is also 

ELEC__KW - I think this reports the set size of the Electric Heat System

INDOORKW - Not sure - Mine comes up as 2413.57251

REHEATKW - [from the latest firmware, I don't know what it does since I haven't updated my thermostat and likely never will]

BLWRWATT - Reports the current wattage the blower motor [NOT the inducer motor] is consuming
ECM_WATT -  Not Sure, but this Tracks almost linearly with BLWRWATT for the short time I've watched it.

jdanders, 2025-03-19:
I've got a dual-fuel furnace. Indoor KW is not reporting anything for me (0 value), while ODU KW seems to be active.