esphome-econet/esphome-econet #232: Add Software Versions, Model #s, and Serial #s for Furnace & ODUs

justbruce, 2023-12-03:
The currently displayed Software Version in Home Assistant and the logs is the Thermostat.

  It would be convenient if the ESPHome device could display the Software Versions for all the devices in the system, in my case the Furnace and the Heat Pump.

  Also in the convenience and troubleshooting category would be retrieving and displaying the Model and Serial #s for each of the devices, too.

It seems like this might be possible, as the Thermostat seems to be able to display all of this without having it entered by the installer or user.

justbruce, 2023-12-05:
I think the appropriate data points are 

HRHEEMID  - Furnace Model No & Serial No 

HWELL_ID - Furnace SW Version

PRODMODN - Heat Pump Model No

PRODSERN - Heat Pump Serial Number

The data shows up in one of my logs [where I pulled the thermostat then plugged it back in] but not when I add it to the yaml - I wonder if we aren't putting proper DST  or SRC addresses  when requesting this data?

stockmopar, 2023-12-06:
I'm pretty sure the Serial Number for Furnaces is in "SERIAL_N" just don't know how to decode it yet.

justbruce, 2023-12-07:
> I'm pretty sure the Serial Number for Furnaces is in "SERIAL_N" just don't know how to decode it yet.

I have yet to see SERIAL_N in my logs.  Will keep looking, but I know for a fact it is part of HREEMID. [the second part, to be precise]

tronikos, 2023-12-07:
See https://github.com/esphome-econet/esphome-econet/wiki/Objects#home-assistant-services how you can try out SERIAL_N

justbruce, 2023-12-08:
> See https://github.com/esphome-econet/esphome-econet/wiki/Objects#home-assistant-services how you can try out SERIAL_N

Nothing returned for SERIAL_N

tronikos, 2023-12-08:
Not even UNSUPPORTED?

justbruce, 2023-12-08:
> Not even UNSUPPORTED?

not even that

stockmopar, 2023-12-08:
So @justbruce is spot on that requests that we define in these ways would target the thermostat not the furnace.

stockmopar, 2023-12-08:
I'll see if I can find HRHEEMID in my logs and/or try requesting them.

justbruce, 2023-12-08:
[justbruce-hvac-log-4.txt](https://github.com/esphome-econet/esphome-econet/files/13617568/justbruce-hvac-log-4.txt)

stockmopar, 2023-12-11:
This doesn't fully support this... but this: https://github.com/esphome-econet/esphome-econet/pull/250 does allow you to request this data point using a Home Assistant service.

stockmopar, 2023-12-11:
Updated https://github.com/esphome-econet/esphome-econet/pull/250 with working example (at least on my system) to parse this when requested from home assistant service.

justbruce, 2023-12-11:
> > Maybe do it on a per-device basis?  I've been kinda wondering if it doesn't make more sense to monitor and report on each device in turn [furnace, heat pump, air handler, thermostat, etc] that is on the RS485 line.  Again, not sure if this is fully fleshed, but I've been pondering it
> 
> Can you explain what you mean by `monitor and report on each device in turn`?
> 
> What does the outcome look like that you're referring to?

At present, the whole system is presented as a sorta monolithic blob, with data points from every device mixed in amongst each other, rather than being separated out.  

  For me, it would be helpful to see all the furnace entities together, and all the heat pump entites together, with the ability to query those items more or less frequently depending on the actual unit they are a part of.  

 I think the work @stockmopar is doing with request_mod intervals and specifying addresses gets most of this(eventually) , but grouping them within Home Assistant in separate boxes, per Device, would be a very useful thing, too.  

Basically, i'd like to stop treating the whole system as a monolith, recognizing that each device is separate

stockmopar, 2023-12-11:
ESPHome doesn't really have a way to make a singular ESPHome device create or spawn sub-devices that are separated or self-contained as we would like in this regard. To do this we would end up essentially needing to change to a Home Assistant integration scheme and the ESPHome device would shuttle all reads / writes back and forth.

justbruce, 2023-12-12:
> ESPHome doesn't really have a way to make a singular ESPHome device create or spawn sub-devices that are separated or self-contained as we would like in this regard. To do this we would end up [...]

OK.  Interesting, if slightly annoying.  As I said, not familiar anough as yet, so thanks for the education.  HA Integration is a BIG project, so I think I'll just content myself with making cards that separate the entities for ease of browsing.....

stockmopar, 2023-12-13:
This PR (https://github.com/esphome-econet/esphome-econet/pull/253) adds the base functionality for supporting this and an example for adding this to a yaml. A future commit could be added to put the example into econet_hvac_furnace.yaml.

closed 2023-12-23 by stockmopar