esphome-econet/esphome-econet #236: Hotwater / Vacatime are unsupported - Gen3 Heatpump

Ylianst, 2023-12-04:
In looking at my Rheem Heatpump Water Header on ESPHOME-ECONET logs, I am seeing these errors periodically.

[image]

The two important lines are here:

```
[23:22:09][W][econet:337]:   HOTWATER : UNSUPPORTED
[23:22:09][W][econet:337]:   VACATIME : UNSUPPORTED
```

If I recall correctly, I have a Gen3 (with the color LCD display). The integration seems to work well otherwise, but if there is a way to remove these errors, it would be great.

Thanks for the wonderful work!
Ylian

tronikos, 2023-12-04:
See https://github.com/esphome-econet/esphome-econet/issues/203#issuecomment-1827130082
I guess we need to add a FAQ and mention this there.

barndawgie, 2023-12-04:
Once https://github.com/esphome/esphome/pull/5821 is merged it should provide a good option for removing components that aren't supported on certain devices.

closed 2023-12-05 by Ylianst

Ylianst, 2023-12-05:
Nice. Indeed this fixed it. Here is my new configuration file:

```
---
substitutions:
  tx_pin: GPIO19
  rx_pin: GPIO22
  platform: esp32
  board: esp32dev

packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    file: econet_heatpump_water_heater.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hotwaterheater Fallback Hotspot"
    password: "xxxxxxxxxxxxxx"

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxx"

sensor:
  - id: !extend hot_water
    request_mod: none
    internal: true
number:
  - id: !extend vacatime
    request_mod: none
    internal: true
```

I added the 8 last lines. It correctly removed the two features:

[image]