esphome-econet/esphome-econet #240: Cooling stage displays Compressor stage when ODU is a Heat Pump

justbruce, 2023-12-06:
Cooling Stage is a bit of a misnomer when the [communicating, in my case] Heat Pump is on in the cooler months.

  Perhaps utilize the !remove directive (when it becomes available) to write build-yaml files that change the terminology based on a customer's configured equipment?

justbruce, 2023-12-06:
> What is it displaying and what "should" it be showing in the case of the ODU being a heat pump?

It is displaying the stage the compressor is in, regardless of Heating or Cooling.  My Heat pump has 3 stages, so just changing the displayed name to "Heat Pump Compressor Stage" would be truer and less confusing.

stockmopar, 2023-12-06:
Maybe just compressor stage to cover both (A/C and heat pumps). There must be another signal inside HWSTATUS that indicates heating vs cooling for heat pump. If you can get the values of the HWSTATUS message in those 3 states (and what state its actually in) we can try to see if we can determine which signal it is.

justbruce, 2023-12-06:
> Maybe just compressor stage to cover both (A/C and heat pumps). There must be another signal inside HWSTATUS that indicates heating vs cooling for heat pump. If you can get the values of the [...]

Indeed.  One of the things I most desire right now is a status message in HA telling me whether the heat pump or the furnace is providing the heat. 

Which 3 states are you asking for the values of ?  The three stages of the heat pump?  OR heating, cooling, and off?