esphome-econet/esphome-econet #241: Utilize LED on the M5 to Indicate Power / Activity

stockmopar, 2023-12-06:


stockmopar, 2023-12-06:
Looks like here's an example of controlling the LED:

https://github.com/m5stack/M5Atom/blob/00f5a14e1cf6c25d16a4dd011a5f971bca29ec43/examples/Basics/FactoryTest/ATOM_LITE/ATOM_LITE.ino#L4

stockmopar, 2023-12-06:
This page:
https://community.home-assistant.io/t/atom-lite-rgb-led-turns-on-by-itself/452852/6

Gives this yaml to try:
```
light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: 27
    num_leds: 1
    rmt_channel: 0
    chipset: SK6812
    id: status_led
    name: ${name} Light
    effects:
      - random:
      - flicker:
      - addressable_rainbow:
```

barndawgie, 2023-12-06:
This is my Atom Lite config:
```
---
binary_sensor:
  - platform: gpio
    id: atom_button
    pin:
      number: GPIO39
      inverted: true
    name: "ATOM Lite Button"
    icon: "mdi:gesture-tap-button"
    entity_category: "config"
    disabled_by_default: true

light:
  - platform: esp32_rmt_led_strip
    name: "ATOM Lite LED"
    id: atom_led
    pin: GPIO27
    chipset: SK6812
    num_leds: 1
    rmt_channel: 0
    rgb_order: GRB
    icon: "mdi:led-off"
    entity_category: "config"
    disabled_by_default: true

output:
  - platform: template
    id: atom_led_output
    type: binary
    write_action:
      if:
        condition:
          lambda: return state;
        then:
          - light.turn_on: atom_led
        else:
          - light.turn_off: atom_led

# esp32_improv:
#   authorizer: atom_button
#   status_indicator: atom_led_output
```

I haven't been able to get the BLE Improv working. Or rather, it works, but results in repeated wi-fi disconnects that I haven't been able to troubleshoot. But the light and button for sure work. I haven't figured out how to control the LED color yet, though. Maybe those "effects" are what I need.

barndawgie, 2023-12-06:
What exactly do we think it would look like to "use" this LED? GIve wi-fi connectivity status? Successful comms with the Rheem unit in the last X minutes?

barndawgie, 2023-12-07:
> ESPHome does not let you enable the button in HA if "entity_category:" is set to "config", but it works if you either do not set the "entity_category:" or set it as: _entity_category: "diagnostic"_

I'm not sure what you mean. It shows up for me in the config section (as a binary sensor).

barndawgie, 2023-12-09:
> > > ESPHome does not let you enable the button in HA if "entity_category:" is set to "config", but it works if you either do not set the "entity_category:" or set it as: _entity_category: "diagnostic"_
> > 
> > 
> > I'm not sure what you mean. It shows up for me in the config section (as a binary sensor).
> 
> I'm on HA 2023.12 and the latest version of ESPHome and the button shows up under config when it is not enabled which is the default. If you try to enable it, it throws an error in the HA logs stating that you cannot enable that component under config and the button goes away. Note the last line in the error log output below:
> 
> ```
> Source: helpers/entity_platform.py:509
> Integration: Binary sensor ([documentation](https://www.home-assistant.io/integrations/binary_sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+binary_sensor%22))
> First occurred: 6:57:44 PM (1 occurrences)
> Last logged: 6:57:44 PM
> 
> Error adding entities for domain binary_sensor with platform esphome
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 509, in async_add_entities
>     await asyncio.gather(*tasks)
>   File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 754, in _async_add_entity
>     await entity.add_to_platform_finish()
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in add_to_platform_finish
>     await self.async_internal_added_to_hass()
>   File "/usr/src/homeassistant/homeassistant/components/binary_sensor/__init__.py", line 198, in async_internal_added_to_hass
>     raise HomeAssistantError(
> homeassistant.exceptions.HomeAssistantError: Entity binary_sensor.esphome_hphw_gen4_atom_lite_button cannot be added as the entity category is set to config
> ```

I'm almost certain this _was_ working for me, but now I get the error that you mention. I believe the issue is that config entities are meant to be ones that you can use to change device behavior (i.e. not just read-only types like `binary_sensor`).

I think the "fix" would be to expose the button as a button to Home Assistant.

barndawgie, 2023-12-19:
I've added an example ATOM Lite config to the wiki. GIven that we want to support multiple device types, I'm not sure it makes sense to build this into our configs.

closed 2023-12-19 by barndawgie