esphome-econet/esphome-econet #255: Components not working with version greater than 1.5.0

vangorra, 2023-12-13:
My config has remained unchanged but recently decided to update my device. I found that if I install any version newer than 1.5.0, none of the esphome components work. 

What does work (with version > 1.5.0):
- OTA - Can push updates without issue.
- Ping - Can ping the device.
- Connection - UNIFI controller shows the device disconnects and reconnects promptly after an OTA update.

What does not work (with version >1.5.0):
- Web server
- API
- Logs

Hardware:
Using the recommended M5 stack components in the wiki.

Config:
```yaml
---
substitutions:
  name: water_heater
  tx_pin: GPIO19
  rx_pin: GPIO22
  platform: esp32
  board: esp32dev

packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: v1.5.0 # This is where I change the version number.
    file: econet_electric_tank_water_heater.yaml

dashboard_import:
  package_import_url: github://esphome-econet/esphome-econet/build-yaml/${name}-${platform}.yaml@${github_ref}
  import_full_config: false

# Uncomment the below to use Wi-Fi settings from your secrets.yaml file
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
  encryption:
    key: !secret encryption_key

web_server:
  port: 80
```

Versions attempted (but all show the same issue):
- 1.5.1
- 1.6.0
- 1.7.0
- 1.7.1

stockmopar, 2023-12-13:
I can confirm that adding web_server also causes the same for me. Is there a reason you have that enabled?
```
web_server:
  port: 80
```

barndawgie, 2023-12-14:
Does it also fail if you remove the version ref and just pull main?

You are forcing config 1.5.0 but may still be pulling code from main.

vangorra, 2023-12-14:
I ran the latest without web server. It takes a minute but it eventually starts working. I've read using too many components exceeds memory capacity, perhaps that's what's happening here.

barndawgie, 2023-12-17:
Since this doesn't seem to reproduce for others and is resolved by moving  to latest code, I'm resolving.

closed 2023-12-17 by barndawgie