esphome-econet/esphome-econet #266: Support for Filter Reset on Furnaces

stockmopar, 2023-12-16:
Alarm reset did not work to clear / reset the filter reminder for my system (Thermostat + Furnace).

Below is a log of when I used the thermostat to reset it:

```
[06:57:01][I][econet:161]: <<< 80.00.03.40.00.80.00.03.80.00.DC.00.00.06.07.80.00.00.3F.80.00.00.34.81.00.00.53.30.30.31.5F.53.20.43.68.61.6E.67.65.20.41.69.72.20.46.69.6C.74.65.72.20.52.65.6D.69.6E.64.65.72.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.F7.89 (236)
[06:57:01][I][econet:162]:   Dst Adr : 0x340
[06:57:01][I][econet:163]:   Src Adr : 0x380
[06:57:01][I][econet:164]:   Command : 6
[06:57:01][I][econet:165]:   Data    : 07.80.00.00.3F.80.00.00.34.81.00.00.53.30.30.31.5F.53.20.43.68.61.6E.67.65.20.41.69.72.20.46.69.6C.74.65.72.20.52.65.6D.69.6E.64.65.72.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00 (220)
[06:57:01][I][econet:306]:   ALRMALRT : 1.000000
[06:57:01][I][econet:314]:   ALARM_01 : (S001_S Change Air Filter Reminder)
[06:57:01][I][econet:314]:   ALARM_02 : ()
[06:57:01][I][econet:314]:   ALARM_03 : ()
[06:57:01][I][econet:314]:   ALARM_04 : ()
[06:57:01][W][component:214]: Component econet took a long time for an operation (0.18 s).
[06:57:01][W][component:215]: Components should block for at most 20-30ms.
[06:57:01][I][econet:386]: Read 138. ms=171355093
[06:57:01][I][econet:161]: <<< 80.00.03.80.00.80.00.03.40.00.7A.00.00.1E.02.01.00.00.56.41.43.41.45.4E.41.42.00.00.56.41.43.53.54.41.54.45.00.00.43.4F.4F.4C.56.41.43.41.00.00.48.45.41.54.56.41.43.41.00.00.56.41.43.41.53.54.49.4D.00.00.56.41.43.41.45.54.49.4D.00.00.56.41.43.41.53.44.41.59.00.00.56.41.43.41.45.44.41.59.00.00.56.41.43.41.53.4D.4F.4E.00.00.56.41.43.41.45.4D.4F.4E.00.00.56.41.43.41.53.59.4F.43.00.00.56.41.43.41.45.59.4F.43.97.BE (138)
[06:57:01][I][econet:162]:   Dst Adr : 0x380
[06:57:01][I][econet:163]:   Src Adr : 0x340
[06:57:01][I][econet:164]:   Command : 30
[06:57:01][I][econet:165]:   Data    : 02.01.00.00.56.41.43.41.45.4E.41.42.00.00.56.41.43.53.54.41.54.45.00.00.43.4F.4F.4C.56.41.43.41.00.00.48.45.41.54.56.41.43.41.00.00.56.41.43.41.53.54.49.4D.00.00.56.41.43.41.45.54.49.4D.00.00.56.41.43.41.53.44.41.59.00.00.56.41.43.41.45.44.41.59.00.00.56.41.43.41.53.4D.4F.4E.00.00.56.41.43.41.45.4D.4F.4E.00.00.56.41.43.41.53.59.4F.43.00.00.56.41.43.41.45.59.4F.43 (122)
[06:57:01][I][econet:180]:   Type    : 2
[06:57:01][I][econet:181]:   PropType: 1
[06:57:01][I][econet:198]:   VACAENAB
[06:57:01][I][econet:198]:   VACSTATE
[06:57:01][I][econet:198]:   COOLVACA
[06:57:01][I][econet:198]:   HEATVACA
[06:57:01][I][econet:198]:   VACASTIM
[06:57:01][I][econet:198]:   VACAETIM
[06:57:01][I][econet:198]:   VACASDAY
[06:57:01][I][econet:198]:   VACAEDAY
[06:57:01][I][econet:198]:   VACASMON
[06:57:01][I][econet:198]:   VACAEMON
[06:57:01][I][econet:198]:   VACASYOC
[06:57:01][I][econet:198]:   VACAEYOC
[06:57:01][W][component:214]: Component econet took a long time for an operation (0.14 s).
[06:57:01][W][component:215]: Components should block for at most 20-30ms.
[06:57:01][I][econet:386]: Read 185. ms=171355264
[06:57:02][I][econet:161]: <<< 80.00.03.40.00.80.00.03.80.00.62.00.00.06.08.82.00.00.00.03.4E.6F.20.08.82.00.00.00.03.4E.6F.20.07.80.00.00.42.AA.00.00.07.80.00.00.42.78.00.00.07.80.00.00.42.A8.00.00.07.80.00.00.42.A8.00.00.07.80.00.00.41.88.00.00.07.80.00.00.41.88.00.00.07.80.00.00.40.A0.00.00.07.80.00.00.40.A0.00.00.07.80.00.00.41.B8.00.00.07.80.00.00.41.B8.00.00.D8.3D (114)
[06:57:02][I][econet:162]:   Dst Adr : 0x340
[06:57:02][I][econet:163]:   Src Adr : 0x380
[06:57:02][I][econet:164]:   Command : 6
[06:57:02][I][econet:165]:   Data    : 08.82.00.00.00.03.4E.6F.20.08.82.00.00.00.03.4E.6F.20.07.80.00.00.42.AA.00.00.07.80.00.00.42.78.00.00.07.80.00.00.42.A8.00.00.07.80.00.00.42.A8.00.00.07.80.00.00.41.88.00.00.07.80.00.00.41.88.00.00.07.80.00.00.40.A0.00.00.07.80.00.00.40.A0.00.00.07.80.00.00.41.B8.00.00.07.80.00.00.41.B8.00.00 (98)
[06:57:02][I][econet:332]:   VACAENAB : 0 (No)
[06:57:02][I][econet:332]:   VACSTATE : 0 (No)
[06:57:02][I][econet:306]:   COOLVACA : 85.000000
[06:57:02][I][econet:306]:   HEATVACA : 62.000000
[06:57:02][I][econet:306]:   VACASTIM : 84.000000
[06:57:02][I][econet:306]:   VACAETIM : 84.000000
[06:57:02][I][econet:306]:   VACASDAY : 17.000000
[06:57:02][I][econet:306]:   VACAEDAY : 17.000000
[06:57:02][I][econet:306]:   VACASMON : 5.000000
[06:57:02][I][econet:306]:   VACAEMON : 5.000000
[06:57:02][I][econet:306]:   VACASYOC : 23.000000
[06:57:02][I][econet:306]:   VACAEYOC : 23.000000
[06:57:02][W][component:214]: Component econet took a long time for an operation (0.13 s).
[06:57:02][W][component:215]: Components should block for at most 20-30ms.
[06:57:02][I][econet:386]: Read 138. ms=171355703
[06:57:02][I][econet:161]: <<< 80.00.03.40.00.80.00.03.80.00.7A.00.00.1E.02.01.00.00.53.57.5F.56.45.52.53.4E.00.00.41.57.5F.53.57.56.45.52.00.00.41.57.4D.41.43.41.44.52.00.00.57.46.53.49.47.4E.41.4C.00.00.57.4C.41.4E.53.54.41.54.00.00.57.4C.53.54.41.54.55.53.00.00.41.57.55.50.47.53.54.41.00.00.53.52.56.52.5F.55.52.4C.00.00.41.57.55.50.47.54.59.50.00.00.41.57.55.50.47.5F.45.4E.00.00.41.4E.54.45.4E.4E.41.00.00.00.43.41.4E.54.45.4E.4E.41.86.68 (138)
[06:57:02][I][econet:162]:   Dst Adr : 0x340
[06:57:02][I][econet:163]:   Src Adr : 0x380
[06:57:02][I][econet:164]:   Command : 30
[06:57:02][I][econet:165]:   Data    : 02.01.00.00.53.57.5F.56.45.52.53.4E.00.00.41.57.5F.53.57.56.45.52.00.00.41.57.4D.41.43.41.44.52.00.00.57.46.53.49.47.4E.41.4C.00.00.57.4C.41.4E.53.54.41.54.00.00.57.4C.53.54.41.54.55.53.00.00.41.57.55.50.47.53.54.41.00.00.53.52.56.52.5F.55.52.4C.00.00.41.57.55.50.47.54.59.50.00.00.41.57.55.50.47.5F.45.4E.00.00.41.4E.54.45.4E.4E.41.00.00.00.43.41.4E.54.45.4E.4E.41 (122)
[06:57:02][I][econet:180]:   Type    : 2
[06:57:02][I][econet:181]:   PropType: 1
[06:57:02][I][econet:198]:   SW_VERSN
[06:57:02][I][econet:198]:   AW_SWVER
[06:57:02][I][econet:198]:   AWMACADR
[... 128 lines cut ...]
[06:57:06][I][econet:198]:   ALRMALRT
[06:57:06][W][component:214]: Component econet took a long time for an operation (0.10 s).
[06:57:06][W][component:215]: Components should block for at most 20-30ms.
[06:57:06][I][econet:386]: Read 311. ms=171359602
[06:57:06][I][econet:161]: <<< 80.00.03.40.00.80.00.03.80.00.DC.00.00.06.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.07.80.00.00.00.00.00.00.DD.7C (236)
[06:57:06][I][econet:162]:   Dst Adr : 0x340
[06:57:06][I][econet:163]:   Src Adr : 0x380
[06:57:06][I][econet:164]:   Command : 6
[06:57:06][I][econet:165]:   Data    : 34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20. [... cut ...]
[06:57:06][W][component:215]: Components should block for at most 20-30ms.
[06:57:07][I][econet:386]: Read 26. ms=171360666
[06:57:07][I][econet:161]: <<< 80.00.00.F1.00.80.00.03.80.00.0A.00.00.1F.07.01.06.31.34.06.10.0C.07.E7.53.FE (26)
[06:57:07][I][econet:162]:   Dst Adr : 0xf1
[06:57:07][I][econet:163]:   Src Adr : 0x380
[06:57:07][I][econet:164]:   Command : 31
[06:57:07][I][econet:165]:   Data    : 07.01.06.31.34.06.10.0C.07.E7 (10)
[06:57:07][I][econet:252]:   ClssType: 7
[06:57:07][I][econet:279]:   DateTime: 2023/12/16 06:52:49.06
```

stockmopar, 2023-12-24:
I'm actually not sure that this is possible to do. This may be internal to the thermostat only. I believe the HCOMMAND above is only the command to the furnace to "do nothing".

justbruce, 2024-02-02:
Can it be reset from the App?  That would tell us if it is possible to remote reset it......

justbruce, 2024-04-02:
> Alarm reset did not work to clear / reset the filter reminder for my system (Thermostat + Furnace).
> 
> Below is a log of when I used the thermostat to reset it:
> 
> ```
> [06:57:01][I][econet:161]: <<< 80.00.03.40.00.80.00.03.80.00.DC.00.00.06.07.80.00.00.3F.80.00.00.34.81.00.00.53.30.30.31.5F.53.20.43.68.61.6E.67.65.20.41.69.72.20.46.69.6C.74.65.72.20.52.65.6D.69.6E.64.65.72.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.F7.89 (236)
> [06:57:01][I][econet:162]:   Dst Adr : 0x340
> [06:57:01][I][econet:163]:   Src Adr : 0x380
> [06:57:01][I][econet:164]:   Command : 6
> [06:57:01][I][econet:165]:   Data    : 07.80.00.00.3F.80.00.00.34.81.00.00.53.30.30.31.5F.53.20.43.68.61.6E.67.65.20.41.69.72.20.46.69.6C.74.65.72.20.52.65.6D.69.6E.64.65.72.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00.34.81.00.00.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.20.00 (220)
> [06:57:01][I][econet:306]:   ALRMALRT : 1.000000
> [06:57:01][I][econet:314]:   ALARM_01 : (S001_S Change Air Filter Reminder)
> [06:57:01][I][econet:314]:   ALARM_02 : ()
> [06:57:01][I][econet:314]:   ALARM_03 : ()
> [06:57:01][I][econet:314]:   ALARM_04 : ()
> [06:57:01][W][component:214]: Component econet took a long time for an operation (0.18 s).
> [06:57:01][W][component:215]: Components should block for at most 20-30ms.
> [06:57:01][I][econet:386]: Read 138. ms=171355093
> ...
> [06:57:03][I][econet:386]: Read 28. ms=171357041
> [06:57:03][I][econet:161]: <<< 80.00.01.C0.00.80.00.03.80.00.0C.00.00.1E.01.01.00.00.48.57.53.54.41.54.55.53.B9.81 (28)
> [06:57:03][I][econet:162]:   Dst Adr : 0x1c0
> [06:57:03][I][econet:163]:   Src Adr : 0x380
> [06:57:03][I][econet:164]:   Command : 30
> [06:57:03][I][econet:165]:   Data    : 01.01.00.00.48.57.53.54.41.54.55.53 (12)
> [06:57:03][I][econet:180]:   Type    : 1
> [06:57:03][I][econet:181]:   PropType: 1
> [06:57:03][I][econet:198]:   HWSTATUS
> [06:57:03][I][econet:386]: Read 163. ms=171357106
> [06:57:03][I][econet:161]: <<< 80.00.03.80.00.80.00.01.C0.00.93.00.00.06.04.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.00.20.00.00.00.00.00.01.01.00.01.01.00.00.00.00.00.00.00.01.42.04.B0.02.00.00.00.00.27.38.02.B6.FE.70.00.00.00.00.00.00.00.00.00.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.7E.00.12.00.68.00.00.00.27.64.00.05.BD.00.5C.A9.00.00.27.00.35.00.0A.00.96.00.11.7E.00.04.49.00.2A.59.06.62.00.00.00.00.00.00.FE.70.18.0A (163)
> [06:57:03][I][econet:162]:   Dst Adr : 0x380
> [06:57:03][I][econet:163]:   Src Adr : 0x1c0
> [06:57:03][I][econet:164]:   Command : 6
> [06:57:03][I][econet:165]:   Data    : 04.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.00.20.00.00.00.00.00.01.01.00.01.01.00.00.00.00.00.00.00.01.42.04.B0.02.00.00.00.00.27.38.02.B6.FE.70.00.00.00.00.00.00.00.00.00.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.7E.00.12.00.68.00.00.00.27.64.00.05.BD.00.5C.A9.00.00.27.00.35.00.0A.00.96.00.11.7E.00.04.49.00.2A.59.06.62.00.00.00.00.00.00.FE.70(147)
> [06:57:03][W][component:214]: Component econet took a long time for an operation (0.11 s).
> [06:57:03][W][component:215]: Components should block for at most 20-30ms.
> [06:57:03][I][econet:386]: Read 55. ms=171357252
> [06:57:03][I][econet:161]: <<< 80.00.01.C0.00.80.00.03.80.00.16.00.00.1F.01.01.04.07.00.00.48.43.4F.4D.4D.41.4E.44.00.00.01.00.00.00.00.00.A9.62 (38)
> [06:57:03][I][econet:162]:   Dst Adr : 0x1c0
> [06:57:03][I][econet:163]:   Src Adr : 0x380
> [06:57:04][I][econet:164]:   Command : 31
> [06:57:04][I][econet:165]:   Data    : 01.01.04.07.00.00.48.43.4F.4D.4D.41.4E.44.00.00.01.00.00.00.00.00 (22)
> [06:57:04][I][econet:252]:   ClssType: 1
> [06:57:04][I][econet:267]:   HCOMMAND: 00.00.01.00.00.00.00.00 (8)
> [06:57:04][I][econet:161]: <<< 80.00.03.80.00.80.00.01.C0.00.01.00.00.06.00.BF.95 (17)
> [06:57:04][I][econet:162]:   Dst Adr : 0x380
> [06:57:04][I][econet:163]:   Src Adr : 0x1c0
> [06:57:04][I][econet:164]:   Command : 6
> [06:57:04][I][econet:165]:   Data    : 00
> [06:57:04][W][component:214]: Component econet took a long time for an operation (0.07 s).
> [06:57:04][W][component:215]: Components should block for at most 20-30ms.
> [06:57:04][I][econet:386]: Read 28. ms=171358278
> ...
> [06:57:06][I][econet:161]: >>> 80.00.03.80.00.80.00.03.40.00.34.00.00.1E.02.01.00.00.41.4C.41.52.4D.5F.30.31.00.00.41.4C.41.52.4D.5F.30.32.00.00.41.4C.41.52.4D.5F.30.33.00.00.41.4C.41.52.4D.5F.30.34.00.00.41.4C.52.4D.41.4C.52.54.6D.7B (68)
[... cut ...]
> [06:57:07][I][econet:162]:   Dst Adr : 0xf1
> [06:57:07][I][econet:163]:   Src Adr : 0x380
> [06:57:07][I][econet:164]:   Command : 31
> [06:57:07][I][econet:165]:   Data    : 07.01.06.31.34.06.10.0C.07.E7 (10)
> [06:57:07][I][econet:252]:   ClssType: 7
> [06:57:07][I][econet:279]:   DateTime: 2023/12/16 06:52:49.06
> ```

Try writing Zero to the FILTUSED datapoint [it's a float].  Don't know if this will actually reset the Replace Filter Alarm, but seems worth a shot.  I don't currently have that feature turned on in the Thermostat