esphome-econet/esphome-econet #292: Add humidity setpoint control (humidifier) in climate entity

stockmopar, 2023-12-27:
This is a placeholder (and split-off from https://github.com/esphome-econet/esphome-econet/issues/289) for future humidifier support. No one on the project has an HVAC humidifier accessory on their furnace/air handler. If you do and want this feature to be implemented, please feel free to add a comment below.

fireheadman, 2023-12-30:
Here to help on this... 
I have a Aprilaire Steam humidifier connected to my Rheem 98% furnace
It only calls to HUM when needed

let me know what you need

fireheadman, 2023-12-30:
just a screenshot from the econet integration for what it sees on HUM
[image]

stockmopar, 2023-12-30:
Assuming that you also have the official Econet app installed you can share screenshots of the available configuration/control settings of humidity control, and then you can also capture logs while exercising all of those controls. I would imagine there is the ability to turn humidity control on/off and set humidity setpoint. It would also be interesting to know how dehumidification and humidity control interact with each other from the Econet app itself.

fireheadman, 2024-01-03:
[logs_econet-hvac_furnace_logs.txt](https://github.com/esphome-econet/esphome-econet/files/13815022/logs_econet-hvac_furnace_logs.txt)

See logs for HUM, hoping this helps.
let it run for a bit and do see references to "HUM" in there

fireheadman, 2024-01-03:
adding some additional info to this...

```
[04:30:15][C][econet:108]:   Datapoint DHUMENAB: enum value (value: 1 : Yes)
[04:30:15][C][econet:108]:   Datapoint DHUMENAB: enum value (value: 1 : Yes)
[04:30:15][C][econet:100]:   Datapoint DHUMSETP: float value (value: 50.000000)
[04:30:15][C][econet:100]:   Datapoint DHUMSETP: float value (value: 50.000000)

[04:30:16][C][econet:108]:   Datapoint HUMDCNFG: enum value (value: 1 : Yes)
[04:30:16][C][econet:108]:   Datapoint HUMDCNFG: enum value (value: 1 : Yes)
[04:30:16][C][econet:108]:   Datapoint HUMDCTRL: enum value (value: 1 : Heat/Fan)
[04:30:16][C][econet:108]:   Datapoint HUMDCTRL: enum value (value: 1 : Heat/Fan)
[04:30:16][C][econet:100]:   Datapoint HUMDSETP: float value (value: 36.000000)
[04:30:16][C][econet:100]:   Datapoint HUMDSETP: float value (value: 36.000000)
[04:30:16][C][econet:108]:   Datapoint HUMSMART: enum value (value: 1 : Low)
[04:30:16][C][econet:108]:   Datapoint HUMSMART: enum value (value: 1 : Low)

04:30:25][D][sensor:094]: 'Thermostat Current Humidity': Sending state 36.29060 % with 1 decimals of accuracy
[04:30:25][D][climate:396]: 'HVAC' - Sending state:
[04:30:25][D][climate:399]:   Mode: HEAT
[04:30:25][D][climate:407]:   Custom Fan Mode: Automatic
[04:30:25][D][climate:419]:   Current Temperature: 19.71°C
[04:30:25][D][climate:423]:   Target Temperature: Low: 18.89°C High: 25.00°C
[04:30:25][D][climate:428]:   Current Humidity: 36%
[04:30:25][D][climate:431]:   Target Humidity: 50%
[04:30:25][I][econet:333]:   SCHEDULE : 1 (Yes)
[04:30:25][I][econet:307]:   SPT : 67.479980
```

fireheadman, 2024-01-29:
anything I do to assist on this issue?

barndawgie, 2024-02-01:
Another option would be to expose as a Humidifier to Home Assistant, though it's not currently exposed by ESPHome: https://github.com/esphome/feature-requests/issues/1859

justbruce, 2025-03-23:
I have an AprilAire 800 Steam Humidifier connected to my Rheem HVAC System.  Having the Humidifier controls exposed would be helpful.

  There are a number of switches and select-able items (Humidify w/Fan, Smart Humidity Control, Humidify During, etc) - this may warrant it's own sub-section.  (ideally it would warrant it's own sub-card, but wihout us going full-on integration I don't know if that is possible]

The PR for the Humidifiers and Hygrostats in ESPHome is stalled at present [and has been since August of Last year].  Should we just bring it in anyway, then convert over if ever the PR moves forward?

justbruce, 2025-03-24:
#471 Adds some controls for humidification

justbruce, 2025-03-24:
> > anything I do to assist on this issue?
> 
> The best approach is likely for the code to check if `HUMDCNFG` and `HUMDENAB` and the AUTOMODE is heat then show HUMDSETP else DHUMSETP if DHUMENAB and AUTOMODE is cool

HUMDCNFG [This is the enable Datapoint for Humidity - not consistent, I know but there ya go] and DHUMENAB can be set at the same time.   For example I have the humidification target set to 38%, and the Dehumidification set to 50%.  During a 2 week period in the Fall, and I'm betting in Spring, both functions will occasionally cycle on.  Keeps the hardwood happy. :-)

Humidification can take place during a cooling cycle, at least on my thermostat.

Given these 2 things, I'm not sure the above logic makes sense. Or maybe it does, I'm open to being wrong

 Ideally the controls would show a target humidity *window*, since that is what most people want anyway.  But that is beyond my ken to code up at present, and maybe a bit off topic here.....