esphome-econet/esphome-econet #302: Units Not Matching HASS Expectations for Some HVAC Sensors

barndawgie, 2023-12-29:
Discord user `@fireheadman` reported the following in his HASS logs:
```
2023-12-28 14:49:01.075 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_odu_pressure_suction (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'PSI' which is not a valid unit for the device class ('pressure') it is using; expected one of ['hPa', 'Pa', 'bar', 'cbar', 'mmHg', 'inHg', 'mbar', 'psi', 'kPa']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.100 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_flame_sensor (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'uA' which is not a valid unit for the device class ('current') it is using; expected one of ['A', 'mA']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.122 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_furnace_high_heat_btu_output (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'btu/h' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.139 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_furnace_low_heat_btu_output (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'btu/h' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.461 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_instant_btu_output (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kbtu/h' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.480 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_instant_btu_usage (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kbtu/h' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.537 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_indoor_kw (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.537 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_indoor_kw (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['kWh', 'MJ', 'MWh', 'GJ', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
2023-12-28 14:51:14.772 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.econet_hvac_furnace_odu_pressure_suction (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'PSI' which is not a valid unit for the device class ('pressure') it is using; expected one of ['hPa', 'Pa', 'bar', 'cbar', 'mmHg', 'inHg', 'mbar', 'psi', 'kPa']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```

Seems like we may need to update some of our configurations with unit conversions to make sure we are using units listed at https://www.home-assistant.io/integrations/sensor/

barndawgie, 2023-12-29:
@stockmopar - Do you know what the `INDOORKW` datapoint actually measures? If, as the name implies, it's instant KW usage then I think it should be `power` rather than `energy` class (and we can set the units).

stockmopar, 2023-12-29:
Here are my thoughts on these:

- sensor.econet_hvac_furnace_odu_pressure_suction
  -  is using native unit of measurement 'PSI' 
  - which is not a valid unit for the device class ('pressure') it is using; 
  - expected one of ['hPa', 'Pa', 'bar', 'cbar', 'mmHg', 'inHg', 'mbar', 'psi', 'kPa'];
 
```Change to psi```
- sensor.econet_hvac_furnace_flame_sensor
  - is using native unit of measurement 'uA' 
  - which is not a valid unit for the device class ('current') it is using; 
  - expected one of ['A', 'mA'];
```
This one it would actually be nice to get uA added to HA, but I'm good with changing to mA
This will cause the values to always be the range of 0.000mA and 0.030mA.
```
- sensor.econet_hvac_furnace_furnace_high_heat_btu_output 
  - is using native unit of measurement 'btu/h' 
  - which is not a valid unit for the device class ('power') it is using; 
  - expected one of ['kW', 'W']; 
```
This would be nice to get a new device class added into HA or expand the possible power units available (ie add btu/h)
Maybe in the short term we actually just don't classify this as power?
```
- sensor.econet_hvac_furnace_furnace_low_heat_btu_output 
  - is using native unit of measurement 'btu/h' 
  - which is not a valid unit for the device class ('power') it is using; 
  - expected one of ['kW', 'W']; 
```
This would be nice to get a new device class added into HA or expand the possible power units available (ie add btu/h)
Maybe in the short term we actually just don't classify this as power?
```
- sensor.econet_hvac_furnace_instant_btu_output 
  - is using native unit of measurement 'kbtu/h' 
  - which is not a valid unit for the device class ('power') it is using; 
  - expected one of ['kW', 'W']; 
```
This would be nice to get a new device class added into HA or expand the possible power units available (ie add btu/h)
Maybe in the short term we actually just don't classify this as power?
```
- sensor.econet_hvac_furnace_instant_btu_usage  
  - is using native unit of measurement 'kbtu/h' 
  - which is not a valid unit for the device class ('power') it is using; 
  - expected one of ['kW', 'W']; 
```
This would be nice to get a new device class added into HA or expand the possible power units available (ie add btu/h)
Maybe in the short term we actually just don't classify this as power?
```
- sensor.econet_hvac_furnace_indoor_kw 
  - is using state class 'measurement' 
  - which is impossible considering device class ('energy') it is using; 
  - expected None or one of 'total_increasing', 'total'; 
```
This does indeed seem to be energy (kWh)
I believe we should remove measurement and add total_increasing
```
- sensor.econet_hvac_furnace_indoor_kw  
  - is using native unit of measurement 'None' 
  - which is not a valid unit for the device class ('energy') it is using; 
  - expected one of ['kWh', 'MJ', 'MWh', 'GJ', 'Wh']; 
```
This does indeed seem to be energy (kWh)
I believe we should remove measurement and add total_increasing
```
- sensor.econet_hvac_furnace_odu_pressure_suction 
  - is using native unit of measurement 'PSI' 
  - which is not a valid unit for the device class ('pressure') it is using; 
  - expected one of ['hPa', 'Pa', 'bar', 'cbar', 'mmHg', 'inHg', 'mbar', 'psi', 'kPa']; 

```Change to psi```

barndawgie, 2023-12-29:
@stockmopar we _could_ convert btu/h to Watts. Do you prefer not to because Watts isn't really used for measuring gas-based power?

barndawgie, 2023-12-29:
Anyway, I agree with the above as at least short-term fixes so I've pushed them into https://github.com/esphome-econet/esphome-econet/pull/303

barndawgie, 2023-12-29:
What is INDOORKW measuring? The furnace's use of electricity (for fan, etc...)? The heating power usage converted to kWh?

fireheadman, 2023-12-29:
My 10 cents (will may not be valued at 10cents...)
Some of these will be repetitive from @stockmopar comments.

**econet_hvac_furnace.yaml**
This line, should be made as a visible comment and/or maybe a dropdown selector in controls section (some devices can change F to C).... so have it change NG or LP for users to select it.   I wasn't aware of it until I dug into the code.
`furnace_kbtus_per_m3: "88.852" # Natural Gas = approx 35.300, Propane = approx 88.852`

These blocks (roughly line 121-136)
I might be out of line on this.... but changing it to the following seemed to report a better output for me:
changed unit_of_measurement: "BTU/(h⋅ft²)"
changed device_class: "irradiance"
```
  - platform: econet
    name: "Furnace High Heat BTU Output"
    id: furnace_hh_btus
    sensor_datapoint: FURNHBTU
    entity_category: "diagnostic"
    unit_of_measurement: "BTU/(h⋅ft²)"
    device_class: "irradiance"
    request_once: true
  - platform: econet
    name: "Furnace Low Heat BTU Output"
    id: furnace_lh_btus
    sensor_datapoint: FURNLBTU
    entity_category: "diagnostic"
    unit_of_measurement: "BTU/(h⋅ft²)"
    device_class: "irradiance"
    request_once: true
```

Did the similar thing here to (roughly line 264-277)
changed unit_of_measurement: "BTU/(h⋅ft²)"
changed device_class: "irradiance"
```
  - platform: template
    name: "Instant BTU Output"
    id: instant_btu_output
    unit_of_measurement: "BTU/(h⋅ft²)"
    device_class: "irradiance"
    filters:
      - delta: 0.9
  - platform: template
    name: "Instant BTU Usage"
    id: instant_btu_usage
    unit_of_measurement: "BTU/(h⋅ft²)"
    device_class: "irradiance"
    filters:
      - delta: 0.9
```

and for this block:   
changed the state_class: total_increasing 
added unit_of_measurement: kWh
```
  - platform: econet
    name: "Indoor KW"
    id: indoor_kw
    sensor_datapoint: INDOORKW
    entity_category: "diagnostic"
    state_class: total_increasing
    unit_of_measurement: "kWh"
    device_class: energy
```

fireheadman, 2023-12-29:
for the yaml:  econet_hvac_odu.yaml
....the only change was (as @stockmopar mentioned)

Changing PSI to psi.   Guess case sensitivity must be followed.
NOTE:  when I changed this, I had to go into the recorder and "Fix Issue" to change PSI to psi for the recorder DB.   It will bark at you because it can't parse old data tagged with 'PSI' and new data with 'psi'.  Select the option to merge without losing data points... Wish I had taken a screenshot of this...
```
  - platform: econet
    name: "ODU Pressure Suction"
    id: odu_pressure_suction
    sensor_datapoint: PRES_SUC
    request_mod: none
    unit_of_measurement: "psi"
    accuracy_decimals: 1
    device_class: "pressure"
    state_class: "measurement"
    entity_category: "diagnostic"
```

fireheadman, 2023-12-29:
UPDATE on the PSI vs psi for ODU suction
I reverted my code back to the current ESPHome code for HVAC, this triggered the response I mentioned above for parsing/merging the recorder data.

Here is the message from logs:
`2023-12-29 14:20:10.544 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.econet_hvac_furnace_odu_pressure_suction (PSI) cannot be converted to the unit of previously compiled statistics (psi). Generation of long term statistics will be suppressed unless the unit changes back to psi or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this`

And here is the screenshot I missed...
Obviously the direction on data change will be opposite, but this gets you the idea of what has to be done once this change is implemented.
[image]
[image]

PR #303 (2023-12-30): fix(#302): Use HASS-Standard Units on Furnace Entities
Fix #302 but updating/converting some units.

Setting as draft for now pending discussion of a few points in #302.
closed 2023-12-30 by barndawgie