esphome-econet/esphome-econet #305: Enhancement:   Request for LP / NG selector in Controls section

fireheadman, 2023-12-29:
Would be nice to offer a selector on fuel type...
[image]

Played around with this a little, but I'm not savvy enough to get the code correct....
This would populate a selector, but it didn't to my knowledge actaually do anything, hoping maybe someone with more experience can do this?

This was placed in the ESPHome deployment code, saved, then install

```
select:
## furnace_kbtus_per_m3: "88.852" # Natural Gas = approx 35.300, Propane = approx 88.852
  - platform: template
    name: Fuel Type
    id: furnace_kbtus_per_m3
    optimistic: True
    options:
      - NG
      - LP
    icon: "mdi:fuel"
```

justbruce, 2024-02-18:
This requires an ability to set a substitution value based on the value of the select.  Which I cannot seem to find a way to do, despite beating my head against it for 2 days.

  In fact, I don't know how to make anything happen based on the value of a select.  The esphome docs are (very, very) POOR in this case.

fireheadman, 2024-02-22:
I was hoping there was IF THEN ELSE logic available for esphome code...  but this is dependent on how/what executes against the yaml.

There could be the option to use configs for this?
Pulled this from stackoverflow, perhaps this might work?
REF:   https://stackoverflow.com/questions/54194104/if-condition-possible-in-yaml-file

```
An elegant option, when having multiple configuration files would be as follows:

application.yml

kafka.bootstrap-servers-nonprod: "nonprod-value"
kafka.bootstrap-servers-prod: "prod-value"
kafka.bootstrap-servers: ${kafka.bootstrap-servers-${env-type}}
application-qa.yml

env-type: "nonprod"
#kafka.bootstrap-servers will have the value: "nonprod-value"
application-prod.yml

env-type: "prod"
#kafka.bootstrap-servers will have the value: "prod-value"
```

justbruce, 2024-02-23:
I think this needs to be something handled on the backend, in the python or c++ code.  Heck if I could figure out how to get the yaml "select" value to be passed back to a python script, I'd be golden.  But I'm a bit too new (and old) to grok the inner workings quite yet.

barndawgie, 2024-02-23:
I think this is actually pretty straightforward and I will take a look once I'm back from my current holiday.

Basically we create a [Template Selector](https://esphome.io/components/select/template.html) with the options and then add some logic in the lambda to case out the calculation.

fireheadman, 2024-02-23:
I have the template selector in the OP... although it does not have an "initial" value set.    Thanks for wanting to get this rolling.

justbruce, 2024-03-30:
There is a datapoint for this that the Econet Thermostat stores - GAS_TYPE ...

 Not sure if this changes things, helps, or hurts what you are trying to accomplish, but thought I'd share.  It is an Enum_Text type

fireheadman, 2024-04-10:
> There is a datapoint for this that the Econet Thermostat stores - GAS_TYPE ...
> 
> Not sure if this changes things, helps, or hurts what you are trying to accomplish, but thought I'd share. It is an Enum_Text type

@justbruce   Would you happen to have a sample of the logging that shows this "GAS_TYPE" with a value?  I looked at my output from esphome (with debugging enabled) and didn't see anything matching that label.

If there is a flag I need to set to make it appear, please let me know....  I think this would make it easier to use a select to set this.

justbruce, 2024-04-11:
> @justbruce Would you happen to have a sample of the logging that shows this "GAS_TYPE" with a value? I looked at my output from esphome (with debugging enabled) and didn't see anything matching that [...]

`[21:01:25][I][econet:161]: >>> 80.00.03.80.00.80.00.03.40.00.0C.00.00.1E.02.01.00.00.47.41.53.5F.54.59.50.45.89.F1 (28)`
`[21:01:25][I][econet:162]:   Dst Adr : 0x380`
`[21:01:25][I][econet:163]:   Src Adr : 0x340`
`[21:01:25][I][econet:164]:   Command : 30`
`[21:01:25][I][econet:165]:   Data    : 02.01.00.00.47.41.53.5F.54.59.50.45 (12)`
`[21:01:25][I][econet:180]:   Type    : 2`
`[21:01:25][I][econet:181]:   PropType: 1`
`[21:01:25][I][econet:198]:   GAS_TYPE`
`[21:01:25][I][econet:161]: 80.00.03.40.00.80.00.03.80.00.11.00.00.06.10.82.00.00.00.0B.4E.61.74.75.72.61.6C.20.47.61.73.89.6D (33)`
`[21:01:25][I][econet:162]:   Dst Adr : 0x340`
`[21:01:25][I][econet:163]:   Src Adr : 0x380`
`[21:01:25][I][econet:164]:   Command : 6`
`[21:01:25][I][econet:165]:   Data    : 10.82.00.00.00.0B.4`
`[21:01:25][I][econet:333]:   GAS_TYPE : 0 (Natural Gas)`

fireheadman, 2024-04-11:
very odd... I can't find that in my logs.    Unsure if it is related to me being on propane and having to force a different value in my yaml config for "furnace_kbtus_per_m3: "88.852""

what model furnace do you have?

justbruce, 2024-04-11:
> very odd... I can't find that in my logs. Unsure if it is related to me being on propane and having to force a different value in my yaml config for "furnace_kbtus_per_m3: "88.852""
> 
> what model furnace do you have?

You have to ask the thermostat for it via HA Developer Tools - Service, then put ESPHome: econet_hvac_read in the 'Service' text box and 'GAS_TYPE' in the Datapoint text box.  Or, alternatively, put it in as a sensor in your econet_hvac_furnace.yaml

tronikos, 2024-04-11:
See also https://github.com/esphome-econet/esphome-econet/wiki/Objects#home-assistant-services