esphome-econet/esphome-econet #319: Changing the target temperature using an automation (electric usage shifting)

Ylianst, 2024-01-13:
I have a 80 gallon heat-pump water heater and, in order to save energy and/or shift electrics consumption, I want to change the target temperature using a home assistant automation. I can change the value on the dashboard, but not within an automation. There are many actions, but not the temperature one?

[image]

Maybe I am missing something. I could turn off the water heater during peek hours, but best is to lower the temp during that time.

Thanks in advance for any tips and tricks.

barndawgie, 2024-01-13:
ESPHome-EcoNet Water Heaters are actually represented as Climate entities in Home Assistant. Try using the Climate->Set Target Temperature service.

Ylianst, 2024-01-19:
Thanks. I will give it a try.

closed 2024-01-19 by Ylianst