esphome-econet/esphome-econet #325: Not seeing power sensor data on Rheem Hybrid electric water heater

kenyonj, 2024-01-20:
My water heater isn't showing power information. Is it supported with this water heater?

Rheem 50 gal hybrid electric
Model # XE50T10HD50U0

[image]

My config:

```yml
substitutions:
  name: econet-hpwh
  friendly_name: Water Heater
packages:
  esphome-econet.esphome-econet: github://esphome-econet/esphome-econet/build-yaml/econet-hpwh-esp32.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: ...
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

barndawgie, 2024-01-20:
Can you share your logs? Add a substitution for `logger_level: DEBUG`, install, then post a minute or two of log output.

kenyonj, 2024-01-24:
[logs_econet-hpwh_logs.txt](https://github.com/esphome-econet/esphome-econet/files/14031018/logs_econet-hpwh_logs.txt)

Logs are attached ^^

barndawgie, 2024-01-24:
Yes. I think that means your appliance isn't returning any data for the requested datapoint. It's possible it uses a different datapoint for the same data, but you'd have to do some experimentation (e.g. using the Read Service we expose in HASS).

Simpler might be to remove these from your configuration (e.g. adding the `!remove` command to your ESPHome yaml) or just disable them in Home Assistant.

kenyonj, 2024-01-24:
> you'd have to do some experimentation (e.g. using the Read Service we expose in HASS).

thanks for this pointer, I'll take a look!

barndawgie, 2024-02-01:
Closing this for now as it doesn't seem we have anyway to resolve.