esphome-econet/esphome-econet #329: Add additional hardware options and wiring instructions

bitfliq, 2024-01-22:
Suggesting updates shown here for https://github.com/esphome-econet/esphome-econet/wiki/Recommended-Hardware-Purchase-and-Setup-Instructions

https://github.com/bitfliq/esphome-econet-wiki/commit/c8e9589544c28406832b069598a5829d43593a6d

- Modify purchase instructions to have matched cables
- Add more information about the data vs telephone cables
- Add alternative purchase option and direct shopping carts for Digikey and Mouser for one-click purchase.  These are not referral codes.  Note that these options require different pin outs.
- Modify information for pin-connections.

barndawgie, 2024-01-23:
@bitfliq - I've given you write access to the wiki if you want to go and make these changes.

bitfliq, 2024-01-23:
Done with very minor changes to wording (delta here:
https://github.com/esphome-econet/esphome-econet/wiki/Recommended-Hardware-Purchase-and-Setup-Instructions/_compare/6daf0fcf36422db808b539f2c91958d4d149e16c...32885e794dbe995877190c6e4fd57965b006f205 )

closed 2024-01-23 by bitfliq

bitfliq, 2024-01-24:
With further testing I am not able to get the U085 to work and have removed that section (including parts list links) from the wiki.