esphome-econet/esphome-econet #338: Defrost is a binary state on ODUs

justbruce, 2024-02-01:
Defrost is either ON or OFF on the ODUs, but is confusingly displayed as a 1 or a 0.

  This might seem fine to a computer programmer, but for the average Joe, it would be much more helpful if this were displayed as "Defrost - ON"  or "Defrost - OFF" [or maybe even "Defrost cycle - Running/Not Running"

I've skimmed the esphome documentation but nothing jumps out at me.

barndawgie, 2024-02-01:
This can likely be solved by setting the `device_class` to something appropriate like `running` or `heat`: https://www.home-assistant.io/integrations/binary_sensor/#device-class

justbruce, 2024-02-01:
I had to set the sensor to "binary_sensor" too, but this is working.  Will submit PR

closed 2024-02-01 by barndawgie