esphome-econet/esphome-econet #344: HA Read Service No Longer Working

barndawgie, 2024-02-10:
If I manually rqeuest DISCTEMP (which definitely works as a datapoint); I don't see any parsed response in the logs.

All I'm seeing is:
```
[13:20:35][I][econet:161]: >>> 80.00.03.40.00.80.00.03.40.00.0C.00.00.1E.02.01.00.00.44.49.53.43.54.45.4D.50.C4.EA (28)
[13:20:35][I][econet:162]:   Dst Adr : 0x340
[13:20:35][I][econet:163]:   Src Adr : 0x340
[13:20:35][I][econet:164]:   Command : 30
[13:20:35][I][econet:165]:   Data    : 02.01.00.00.44.49.53.43.54.45.4D.50 (12)
[13:20:35][I][econet:180]:   Type    : 2
[13:20:35][I][econet:181]:   PropType: 1
[13:20:35][I][econet:198]:   DISCTEMP
[13:20:36][I][econet:387]: Read 22. ms=188686226
[13:20:36][I][econet:161]: <<< 80.00.00.F1.00.80.00.12.80.00.06.00.01.1F.09.01.00.00.12.80.4E.FA (22)
[13:20:36][I][econet:162]:   Dst Adr : 0xf1
[13:20:36][I][econet:163]:   Src Adr : 0x1280
[13:20:36][I][econet:164]:   Command : 31
[13:20:36][I][econet:165]:   Data    : 09.01.00.00.12.80 (6)
[13:20:36][I][econet:253]:   ClssType: 9
[13:20:36][I][econet:387]: Read 22. ms=188686280
[13:20:36][I][econet:161]: <<< 80.00.00.F1.00.80.00.12.80.00.06.00.01.1F.09.01.00.00.12.80.4E.FA (22)
[13:20:36][I][econet:162]:   Dst Adr : 0xf1
[13:20:36][I][econet:163]:   Src Adr : 0x1280
[13:20:36][I][econet:164]:   Command : 31
[13:20:36][I][econet:165]:   Data    : 09.01.00.00.12.80 (6)
[13:20:36][I][econet:253]:   ClssType: 9
[13:20:41][I][econet:161]: >>> 80.00.12.80.00.80.00.03.40.00.34.00.00.1E.02.01.00.00.41.4C.41.52.4D.5F.30.31.00.00.41.4C.41.52.4D.5F.30.32.00.00.41.4C.41.52.4D.5F.30.33.00.00.41.4C.41.52.4D.5F.30.34.00.00.41.4C.52.4D.41.4C.52.54.A8.87 (68)
[13:20:41][I][econet:162]:   Dst Adr : 0x1280
[13:20:41][I][econet:163]:   Src Adr : 0x340
[13:20:41][I][econet:164]:   Command : 30
[13:20:41][I][econet:165]:   Data    : 02.01.00.00.41.4C.41.52.4D.5F.30.31.00.00.41.4C.41.52.4D.5F.30.32.00.00.41.4C.41.52.4D.5F.30.33.00.00.41.4C.41.52.4D.5F.30.34.00.00.41.4C.52.4D.41.4C.52.54 (52)
[13:20:41][I][econet:180]:   Type    : 2
[13:20:41][I][econet:181]:   PropType: 1
```

barndawgie, 2024-02-10:
I was able to get the read working by using the `read_from_address` service and specifying the address from the `econet_heatpump_water_heater.yaml` file (0x1280) as a decimal (4736). So seems like maybe the normal `econet_read` call is using the wrong address?

justbruce, 2024-02-11:
> I was able to get the read working by using the `read_from_address` service and specifying the address from the `econet_heatpump_water_heater.yaml` file (0x1280) as a decimal (4736). So seems like maybe the normal `econet_read` call is using the wrong address?

Can confirm..  Also [at least on my system], if one has the logs open the `econet_hvac_read` datapoint requested does appear there.  Looks like it just stopped showing up in HA.

PR #358 (2024-02-27): fix: use correct address for homeassistant_read
fixes #344
closed 2024-02-27 by barndawgie