esphome-econet/esphome-econet #346: Support OCOMMAND decoding

justbruce, 2024-02-10:
Raw Data (including hex]

`[econet:161]: <<< 80.00.04.00.00.80.00.03.80.00.12.00.00.1F.01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.4C.E1.06.40.77.D5 (34)
[I][econet:162]:   Dst Adr : 0x400
[I][econet:163]:   Src Adr : 0x380
[I][econet:164]:   Command : 31
[I][econet:165]:   Data    : 01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.4C.E1.06.40 (18)
[I][econet:253]:   ClssType: 1
[I][econet:261]:   OCOMMAND: 117977600.000000
[I][econet:161]: <<< 80.00.03.80.00.80.00.04.00.00.01.00.00.06.01.7E.66 (17)
[I][econet:162]:   Dst Adr : 0x380
I][econet:163]:   Src Adr : 0x400
[I][econet:164]:   Command : 6
[I][econet:165]:   Data    : 01
`

More to come once my ODU changes speeds [usually overnight as temp drops

justbruce, 2024-02-11:
`[I][econet:161]: <<< 80.00.04.00.00.80.00.03.80.00.12.00.00.1F.01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.4D.0F.23.20.0C.A4 (34)`
`[I][econet:162]:   Dst Adr : 0x400`
`[I][econet:163]:   Src Adr : 0x380`
`[I][econet:164]:   Command : 31`
`[I][econet:165]:   Data    : 01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.4D.0F.23.20 (18)`
`[I][econet:253]:   ClssType: 1`
`[I][econet:261]:   OCOMMAND: 150090240.000000`
`[I][econet:387]: Read 17. ms=1119231`
`[I][econet:161]: <<< 80.00.03.80.00.80.00.04.00.00.01.00.00.06.01.7E.66 (17)`
`[I][econet:162]:   Dst Adr : 0x380`
`[I][econet:163]:   Src Adr : 0x400`
`[I][econet:164]:   Command : 6`
`[I][econet:165]:   Data    : 01`

Resulting settled speed :2290 RPM

justbruce, 2024-02-13:
Now I'm very confused - if you do a decimal to hex conversion on the 9 digit number reported for OCOMMAND, the first two couplets convert back to decimal exactly to the settled speed .  For example 

117977600 <<<---- 07 08 32 00  (first two couplets) == 07 08 ---->>> 1800 

So, why convert the float to a decimal, then drop 2 couplets, then convert back to decimal?

barndawgie, 2024-02-13:
Are they two different values? Target and current or something?

justbruce, 2024-02-14:
No.  When converting the value of OCOMMAND to hex, the last 2 couplets **always** come out to "32 00" or (more rarely) "64 00".  

I think that the OCOMMAND is the commanded speed, and the Inverter Crank Speed [ISCSPEED] is the current speed.

justbruce, 2025-03-24:
> No. When converting the value of OCOMMAND to hex, the last 2 couplets **always** come out to "32 00" or (more rarely) "64 00".
> 
> I think that the OCOMMAND is the commanded speed, and the Inverter Crank Speed [ISCSPEED] is the current speed.

Now that I've had some time to mull this over, I think the last two digits may be a bit-mask.  

20        00100000
40        01000000
03        00000011
06        00000110

What they indicate is still a bit of a mystery, thought I have a suspicion they encode the desired ODU Fan speed.