esphome-econet/esphome-econet #347: Multiple zones not recognized

derekatkins, 2024-02-12:
I have a Rheem Econet HVAC system with 4 zones.  I connected ESPHome today and connected to HA, but only one thermostat shows up (climate.rheem_hvac_hvac).
The other Econet integration does show all four zones.
Is there some configuration I can (or need to) make to get the ESPhome to see the zoning?

barndawgie, 2024-02-12:
We recently added multi-zone support, but not yet documentation for it, I don't think!

@bitfliq might be able to help...

derekatkins, 2024-02-12:
Thanks.  If there is anything I can do to test/QA or otherwise help, please let me know.

bitfliq, 2024-02-12:
Hi @derekatkins - check out https://github.com/esphome-econet/esphome-econet/blob/main/hvac-zone-yaml/remote-hvac-example.yaml and see if that works for zones 1-3.  You're the first with zone 4 to test.

I put in a zone4 yaml into this branch (different git repo) - if you test it and it works we can merge to main.  https://github.com/bitfliq/esphome-econet/tree/zones-zone4.  You'll also need to add it as an include in the package.

derekatkins, 2024-02-12:
@bitfliq Stupid question but...  I just installed from the web; I didn't "build" anything locally.  Are there detailed instructions on how to "build" and "update" my installed device?  Do I have to de-deploy it and connect via USB to my laptop again?  Or is there some way to push an update via TCP/IP?

derekatkins, 2024-02-13:
I can confirm that the zones 1-3 work. I did not "install" zones 4-6, so I commented out:
      hvac-zone-yaml/econet_hvac_zone456status.yaml, #include if using zones 4,5, or 6
and
      hvac-zone-yaml/econet_hvac_zone6_electronicbypass.yaml

derekatkins, 2024-02-14:
HI.
I can now confirm that zone 4 is recognized.  However I am still having the issue that zones 2, 3, and 4 all report as "Off".   The "all zones follow zone 1" setting is turned off on my primary thermostat control.  The system is working (zones 2, 3, and 4 have their own temp settings and properly call for heat/cooling).  But ESPHome does not seem to be reporting that properly.

[image]