esphome-econet/esphome-econet #353: Determine Meaning of ODU Type

barndawgie, 2024-02-17:
We should determine the meaning of ODU Type and then spec it as a text_sensor/enum instead of int.

See #345.

justbruce, 2024-02-17:
Please, if you have an ODU, respond with 

1. the value of ODU Type
2. your Model number [used to determing what series of unit] , and
3.  whether it is an Air Conditioner or Heat Pump.

barndawgie, 2024-02-18:
> Please, if you have an ODU, respond with 
> 
> 
> 
> 1. the value of ODU Type
> 
> 2. your Model number [used to determing what series of unit] , and
> 
> 3.  whether it is an Air Conditioner or Heat Pump.

I put a note in Discord asking folks to comment here with that info.

johncarey70, 2024-02-18:
1. ODU Type 0
2. RA18AZ36AJVCA
3. Air Conditioner

franc6, 2024-02-18:
1. ODU Type 0
2. UA2036AJVCB
3. Air Conditioner

justbruce, 2024-02-19:
> We should determine the meaning of ODU Type and then spec it as a text_sensor/enum instead of int.
> 
> See #345.

As the data gathers, what is your suggested method for converting this to a text sensor/enum?
I look at docs and tried to find examples, but nothing jumps out at me.

derekatkins, 2024-02-20:
My ODU Type says "Unknown".  But it's also not "running" per se.
It's an AC.
I'll go outside at some point to determine the model number and update this comment.

justbruce, 2024-02-21:
> My ODU Type says "Unknown". But it's also not "running" per se. It's an AC. I'll go outside at some point to determine the model number and update this comment. ETA: t is a Classic Plus Series, High Efficiency Air Conditioner, Econet Enabled. Model Number RA1760AJ2CB

Is it powered on right now?  I know it is seems a dumb question but lots of folks pull the disconnect on A/C units in the winter.......

derekatkins, 2024-02-21:
> 
> Is it powered on right now? I know it is seems a dumb question but lots of folks pull the disconnect on A/C units in the winter.......

Yes, the breaker is turned on.

justbruce, 2024-02-21:
> > Is it powered on right now? I know it is seems a dumb question but lots of folks pull the disconnect on A/C units in the winter.......
> 
> Yes, the breaker is turned on.

Hmm.  Interesting.  Do you happen to know the Software Version of the Unit?.....

  This is the first one I'm aware of that reports Unknown.

derekatkins, 2024-02-21:
> Hmm. Interesting. Do you happen to know the Software Version of the Unit?.....
> 
> This is the first one I'm aware of that reports Unknown.

According to the data in HA:

[image]

justbruce, 2024-02-21:
Oh..... This looks like one of the new 2 speed units [not the 3 speed that the 17-series used to be].  I wonder if they changed the datapoint when they changed everything else, too.

justbruce, 2024-02-21:
> Oh..... This looks like one of the new 2 speed units [not the 3 speed that the 17-series used to be]. I wonder if they changed this datapoint name of reporting when they changed everything else, too.

derekatkins, 2024-02-21:
Could be.
I posted some verbose data captures in the discord hvac dev channel.
Although I admit those are from before I added the ODU config.  I won't be 
able to acquire another dump until Thursday.

justbruce, 2024-02-21:
> Could be. I posted some verbose data captures in the discord hvac dev channel. Although I admit those are from before I added the ODU config. I won't be able to acquire another dump until Thursday.

I'm not on Discord [by choice], so I guess someone else will have the fun of parsing those..

:-)

fireheadman, 2024-02-21:
See attached for esphome logs (might be something in there....??)
[logs_rheem-hvac_logs.txt](https://github.com/esphome-econet/esphome-econet/files/14361786/logs_rheem-hvac_logs.txt)

Strange I am not seeing the software version for ODU in my entities, could have sworn it was there before.
I do have this though...
[image]

barndawgie, 2024-02-21:
> See attached for esphome logs (might be something in there....??)
> 
> [logs_rheem-hvac_logs.txt](https://github.com/esphome-econet/esphome-econet/files/14361786/logs_rheem-hvac_logs.txt)
> 
> 
> 
> Strange I am not seeing the software version for ODU in my entities, could have sworn it was there before.
> 
> I do have this though...
> 
> Mine is a 2-Stage 5 ton unit
> 
> [image]
> 
> 

Are you seeing ODU Type showing as 4?

fireheadman, 2024-02-21:
applied the latest esphome updates, which prompted to update all devices.
now I see this...

[image]

fireheadman, 2024-02-21:
What is the OCU Type Datapoint called?   I am looking in the logs for "type" and there is a lot of parse through...

justbruce, 2024-02-22:
> **What is the OCU Type Datapoint called?** I am looking in the logs for "type" and there is a lot of parse through...
> 
> ```
> [fireheadman@templar Downloads 5784]$ grep TYPE logs_rheem_hvac.txt | sort -u
> [23:00:53][I][econet:307]:   CMP_TYPE : 0.000000
> [23:00:57][C][econet:100]:   Datapoint CMP_TYPE: float value (value: 0.000000)
> [23:01:04][C][econet.sensor:026]:   Sensor has datapoint ID ODU_TYPE
> [23:01:09][I][econet:198]:   CMP_TYPE
> [23:01:09][I][econet:307]:   CMP_TYPE : 0.000000
> [23:01:17][I][econet:198]:   CMP_TYPE
> [23:01:17][I][econet:307]:   CMP_TYPE : 0.000000
> [23:01:20][I][econet:198]:   ODU_TYPE
> [23:01:26][I][econet:198]:   CMP_TYPE
> [23:01:34][I][econet:307]:   CMP_TYPE : 0.000000
> [23:01:42][I][econet:198]:   CMP_TYPE
> [23:01:42][I][econet:307]:   CMP_TYPE : 0.000000
> [23:01:51][I][econet:198]:   CMP_TYPE
> [23:01:51][I][econet:307]:   CMP_TYPE : 0.000000
> [23:01:52][I][econet:198]:   ODU_TYPE
> [23:01:58][I][econet:198]:   CMP_TYPE
> [23:01:58][I][econet:307]:   CMP_TYPE : 0.000000
> [23:02:06][I][econet:198]:   CMP_TYPE
> [23:02:07][I][econet:307]:   CMP_TYPE : 0.000000
> [23:02:14][I][econet:198]:   CMP_TYPE
> [23:02:15][I][econet:307]:   CMP_TYPE : 0.000000
> [23:02:22][I][econet:198]:   ODU_TYPE
> [23:02:31][I][econet:198]:   CMP_TYPE
> [23:02:31][I][econet:307]:   CMP_TYPE : 0.000000
> [23:02:40][I][econet:307]:   CMP_TYPE : 0.000000
> [23:02:47][I][econet:198]:   CMP_TYPE
> [23:02:47][I][econet:307]:   CMP_TYPE : 0.000000
> [23:02:52][I][econet:198]:   ODU_TYPE
> [23:02:56][I][econet:307]:   CMP_TYPE : 0.000000
> [23:03:04][I][econet:307]:   CMP_TYPE : 0.000000
> [23:03:12][I][econet:198]:   CMP_TYPE
> [23:03:12][I][econet:307]:   CMP_TYPE : 0.000000
> [23:03:20][I][econet:198]:   CMP_TYPE
> [23:03:20][I][econet:307]:   CMP_TYPE : 0.000000
> [23:03:22][I][econet:198]:   ODU_TYPE
> ```
> 
> And if I grep -i type, then I have a lot more
> 
> ```
> [fireheadman@templar Downloads 5786]$ grep -i TYPE logs_rheem_hvac.txt | sort -u | wc -l
>      407
> ```

The ODU Type datapoint is ODU_TYPE.  I think the CMP_TYPE point is for the type of compressor, though every Econet unit Rheem/Ruud sold until very recently was advertised as having a Copeland Scroll compressor, so I haven't worked on putting the Compressor Type in, as it seemed less than useful

justbruce, 2024-02-22:
> Are you seeing ODU Type showing as 4?

That is my reading of the logs as well.  I wonder why it is showing unknown - unless something else is blocking the read?  Maybe in the request_mod group?  Hmm.....

justbruce, 2024-02-22:
> applied the latest esphome updates, which prompted to update all devices. now I see this...
> 
> ```
> [23:00:58][C][econet:104]:   Datapoint ODUSWVER: text value (value: AC-2SHP-03-01-02)
> [23:00:58][C][econet:104]:   Datapoint ODUSWVER: text value (value: AC-2SHP-03-01-02)
> ```
> 
> ![image](https://private-user-images.githubusercontent.com/19847195/306799076-6ac11e6b-7425-4162-be07-e94878fe3e83.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1Nzc1NjQsIm5iZiI6MTcwODU3NzI2NCwicGF0aCI6Ii8xOTg0NzE5NS8zMDY3OTkwNzYtNmFjMTFlNmItNzQyNS00MTYyLWJlMDctZTk0ODc4ZmUzZTgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIyVDA0NDc0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzNTNiNjdhOGMzMzc4ZGFjY2I1YWQzYjEzYWZlNmY5NTQ1MTQ2M2U4MTYxOGZkZGU3ODY3NDVkMDYwZDBjMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rL2yyOFZlCgXMocsaAlLWd1RLF-mo3eGGJ_k-u9RDqg)

Would you mind taking some more logs?  I have a pet theory as to why the Type is showing as UNKNOWN, and your logs after your update might help...

justbruce, 2024-02-22:
> Are you seeing ODU Type showing as 4?

Is there a way to specify 2 addresses for a datapoint to be read from?  

I'm seeing that @fireheadman has an ODU which is at the 0x500 address, and currently the substitution is set to a 0x400 address.  

Is this something we need to split into a separate file?

barndawgie, 2024-02-22:
> Is this something we need to split into a separate file?

For now it's probably best to just put in a substitution. We can add details like this to the wiki if there are known configs required for certain units (at least for now).

derekatkins, 2024-02-22:
@justbruce --
> Is there a way to specify 2 addresses for a datapoint to be read from?
> 
> I'm seeing that @fireheadman has an ODU which is at the 0x500 address, and currently the substitution is set to a 0x400 address.

Where are you seeing that this is the address of his ODU?  It sounds like he has the same outdoor unit that I have.

> 
> Is this something we need to split into a separate file?

justbruce, 2024-02-22:
> @justbruce --
>
>
> Where are you seeing that this is the address of his ODU? It sounds like he has the same outdoor unit that I have.

The logs.  The SRC address for the OCOMMAND is 0x500

derekatkins, 2024-02-22:
@justbruce --

> The logs. The SRC address for the OCOMMAND is 0x500.

Aha.  Looks like the same for me (not surprising since we have the same unit):

```
[11:06:23][I][econet:161]: <<< 80.00.05.00.00.80.00.03.80.00.12.00.00.1F.01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.00.00.00.00.32.73 (34)
[11:06:23][I][econet:162]:   Dst Adr : 0x500
[11:06:23][I][econet:163]:   Src Adr : 0x380
[11:06:23][I][econet:164]:   Command : 31
[11:06:23][I][econet:165]:   Data    : 01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.00.00.00.00 (18)
[11:06:23][I][econet:253]:   ClssType: 1
[11:06:23][I][econet:261]:   OCOMMAND: 0.000000
[11:06:23][I][econet:161]: <<< 80.00.03.80.00.80.00.05.00.00.01.00.00.06.01.BF.AA (17)
[11:06:23][I][econet:162]:   Dst Adr : 0x380
[11:06:23][I][econet:163]:   Src Adr : 0x500
[11:06:23][I][econet:164]:   Command : 6
[11:06:23][I][econet:165]:   Data    : 01
```

> If you're comfortable testing some local changes and then compile/upload, we can see if I'm right.
> in the file econet_hvac_odu.yaml change  `odu_address: "1024"` to  `odu_address: "1280"` then compile and upload

I push changes through HA's EcoHome Add-On.  So I can update my `econet-hvac_furnace.html` config file, but I do not not have good way to update the component files without creating a new git branch that I can point to.  Is there some way to make this change as a substitution in my main config file?  Worst case I could fork/branch/update, but a simple config change would be easier.

derekatkins, 2024-02-22:
For the record, I added `odu_address: "1280"` to the `substitutions` section of my main config file and pushed the update to the my device.  I am now getting more data!  It is now reporting the ODU Model Number, ODU Serial Number, and ODU Type says "0" instead of "unknown".

There are still a bunch of "unknown" and "0" (or 0.0) value ODU items, but fewer than before.

fireheadman, 2024-02-22:
Here is what I am showing for ODU objects reported:
[image]

[image]

barndawgie, 2024-02-22:
Are we pretty confident at this point that Type 0 is AC and Type 1 is Heat Pump? Are there any counter examples or other values?

fireheadman, 2024-02-22:
I would agree with that since having it set is allowing more variables to populate with accurate data.

justbruce, 2024-02-23:
> Are we pretty confident at this point that Type 0 is AC and Type 1 is Heat Pump? Are there any counter examples or other values?

I'm pretty confident.  This is only like 10 data points, and no one else with a heat pump has yet chimed in, but I'm still pretty happy to call Type 0 AC and Type 1 HP.

justbruce, 2024-02-23:
> 1. ODU type 5
> 
>     2. My ODU is from 2017 timeframe doesn't report model number. I know it's an RP20 2.5ton
> 
>     3. Heat Pump
> 
> 
> Software version: AC-VSHP-00-01-41

Do you happen to know what address it is at? [see issue #354 ]

justbruce, 2024-02-23:
> > > 1. ODU type 5
> > >    
> > >    2. My ODU is from 2017 timeframe doesn't report model number. I know it's an RP20 2.5ton
> > >    3. Heat Pump
> > > 
> > > Software version: AC-VSHP-00-01-41
> > 
> > 
> > Do you happen to know what address it is at? [see issue #354 ]
> 
> My address is 0x400

Well there goes my neat little 0-AC, 1-HP theory.........

tcarney, 2025-01-29:
1. `ODU_TYPE : 1`
2. `PRODMODN : RH3VZ4821STACNJ`
3. `PRODDESC : EcoNet Variable/Hybrid HP`

FWIW, I found this document from Rheem, which does show `AC[0] or Heat Pump[1]` as the definition:
https://pts.myrheem.com/DocStore/cad/EB/EB-108590-07.pdf

I also found that url pattern will work for other "data part numbers".  The contractor app loads them or the data point from the document (`MDPRTNUM` in this case).

justbruce, 2025-01-30:
> 1. `ODU_TYPE : 1`
>
>     2. `PRODMODN : RH3VZ4821STACNJ`
>
>     3. `PRODDESC : EcoNet Variable/Hybrid HP`
>
>
> FWIW, I found this document from Rheem, which does show `AC[0] or Heat Pump[1]` as the [...]

This seems useful.  I'll add all this info to the Protocol  Table I'm maintaining.

jdanders, 2025-03-19:
Sorry for the late addition, hopefully still useful:
 ODU_SIZE : 4.000000
 ODU_TYPE : 1.000000
 PRODMODN : (RD17AZ48AJ3NA)
 PRODDESC : (EcoNet Variable/Hybrid HP)
Heat pump

BUT

In my HA history, I see ODU_TYPE also show up as 5.

[image]