esphome-econet/esphome-econet #354: Some ODUs are at different addresses

justbruce, 2024-02-23:
This prevents some of the data points from being read, as the standard odu_address substitution is 0x400.  As we discovered in #353 , some newer units are at 0x500

Is there a way to set up a text sensor to detect the address from some of the traffic on the RS485 line?  That would be a first step towards 'automagically' selecting the correct address.

  Either way, updating the wiki seems appropriate.

derekatkins, 2024-02-23:
I suspect you could leverage something like the OCOMMAND to pull the data out to see it:
```
[11:06:23][I][econet:161]: <<< 80.00.05.00.00.80.00.03.80.00.12.00.00.1F.01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.00.00.00.00.32.73 (34)
[11:06:23][I][econet:162]:   Dst Adr : 0x500
[11:06:23][I][econet:163]:   Src Adr : 0x380
[11:06:23][I][econet:164]:   Command : 31
[11:06:23][I][econet:165]:   Data    : 01.01.00.07.00.00.4F.43.4F.4D.4D.41.4E.44.00.00.00.00 (18)
[11:06:23][I][econet:253]:   ClssType: 1
[11:06:23][I][econet:261]:   OCOMMAND: 0.000000
[11:06:23][I][econet:161]: <<< 80.00.03.80.00.80.00.05.00.00.01.00.00.06.01.BF.AA (17)
[11:06:23][I][econet:162]:   Dst Adr : 0x380
[11:06:23][I][econet:163]:   Src Adr : 0x500
[11:06:23][I][econet:164]:   Command : 6
[11:06:23][I][econet:165]:   Data    : 01
```
Although I don't know how you would leverage that to "update" the setting in the configuration/code?

justbruce, 2024-02-23:
> I suspect you could leverage something like the OCOMMAND to pull the data out to see it:
>
> ```
> [11:06:23][I][econet:161]: <<< [...]

Using OCOMMAND for the data was what I was thinking.  The backend code for datapoints already parses the DST and SRC addresses for display - it is a matter of getting that parsing into a sensor/text sensor/select so it can be used elsewhere.  I don't yet know how to do that trick of turning the displayed-in-logs info into available-to-use info, unfortunately.

justbruce, 2024-03-30:
> Using OCOMMAND for the data was what I was thinking. The backend code for datapoints already parses the DST and SRC addresses for display - it is a matter of getting that parsing into a sensor/text [...]

so it turns out that there is already a handy Datapoint [assuming one is using an Econet Thermostat] -  ODUADDRS

I'm still not clear how to set the substitution equal to this value, as my brain still considers the whole yaml-and-lambdas thing a bit foreign and byzantine.

barndawgie, 2024-12-24:
Closing this as it seems to have idled for a long time. Feel free to re-open if this issue is still affecting you, @justbruce

closed 2024-12-24 by barndawgie

justbruce, 2025-03-05:
> Closing this as it seems to have idled for a long time. Feel free to re-open if this issue is still affecting you, [@justbruce](https://github.com/justbruce)

please see #459, as this is related.  May be worth reopening this issue, or not.