esphome-econet/esphome-econet #365: Reliability issue with ESP8266

tronikos, 2024-03-10:
The recent changes that added the date and time data points seem to have caused issues with ESP8266. All sensors are unknown, device keeps getting disconnected and I'm unable to even access logs. 2 other users at discord have reported similar issues.

My current working config is:

```yaml
improv_serial: !remove
captive_portal: !remove

sensor:
  - id: !remove wifi_signal_sensor
text_sensor:
  - id: !remove irtc_time
  - id: !remove model_number
  - id: !remove serial_number
  - id: !remove sw_version
  - id: !remove alarm_history_3
  - id: !remove alarm_history_4
number:
  - id: !remove irtc_year
  - id: !remove irtc_month
  - id: !remove irtc_day
  - id: !remove irtc_hour
  - id: !remove irtc_minute
  - id: !remove irtc_seconds
  - id: !remove vacatime
switch:
  - id: !remove alarm_beep
  - id: !remove screen_lock
  - id: !remove display_unit
select:
  - id: !extend vaca_net
    options:
      1: !remove
time:
  - id: !remove homeassistant_time
```

- I have removed all the irtc* that for some reason started causing issues
- I have removed improv_serial and captive_portal to save on resources
- I have removed a few more sensors I don't personally use to limit the periodic read requests
- I have removed the non working vacation option and its timer

The root cause is unknown. Inability to access logs makes this hard to debug. What I know is adding back the irtc* is enough to start having the reliability issues.

@barndawg at discord had the following suggestions:

- Move the clock polling to its own YAML file and don't include it in the ESP8266 configs
- Disable the clock polling in the ESP8266 YAML files
- Significantly increase the polling frequency of the clock (e.g. every hour) as it's primarily just to make sure that the time isn't way off of current time
- Eliminate the polling entirely now that we have the clock syncing

Unfortunately I don't have the time to investigate further.

tronikos, 2024-03-10:
For the record I'm using a D1 Mini ESP8266

dewoodruff, 2024-03-10:
I am using an ESP8266 12-E NodeMCUv2 clone from ebay and ran into the same issues. Removing irtc_*, improv_serial, and captive_portal increased the reliability for me

justbruce, 2024-03-11:
Try adding a request_mod.  

I've noticed [during testing my own stuff] that lots of things seem to block further execution if they are unknown or otherwise wonky - I've seen nearly every sensor come up unknown as well - but if you "isolate" stuff in it's own request_mod group, this behavior is then limited, seemingly.

  Haven't the knowledge to run this down more fully, so think of this as more like a shot-in-the-dark than learned testimony

closed 2024-03-28 by tronikos