esphome-econet/esphome-econet #366: IRTC Clock Sensor Not Following Daylight Savings Time

barndawgie, 2024-03-10:
After DST last night, my cock shows the correct time but the wrong UTC offset (still shows UTC-8 when it is now UTC-7).

barndawgie, 2024-03-10:
Yes - I believe the issue is that the heater has no sense of DST. I can hack in some logic or we can go back to storing UTC time on the heater.

barndawgie, 2024-03-10:
That was my original plan. We have DST mostly working, though, other than the rendering of that string to include the correct time zone. I'm pretty sure I can fix that. It does mean we are setting the clock forward an hour in the spring and back an hour in the fall like an analog clock...

justbruce, 2025-03-11:
> Yes - I believe the issue is that the heater has no sense of DST or timezone - it's like the clock on a microwave. I can hack in some logic or we can go back to storing UTC time on the heater.

The Thermostat has a Data Point which might help you - 2 in fact.  

`Datapoint DST_ENAB: enum value (value: 1 : Yes)`    -      Is Daylight Savings Enabled?
`Datapoint DST_DONE: enum value (value: 1 : Yes) ` -       Are we currently in Daylight Savings Time?

You should be able to check these Data Points, then display your time and UTC offset correctly