esphome-econet/esphome-econet #379: m5 stack boot looping and esphome version missing some details.

Phalen, 2024-03-29:
I jsut completed assembally of the c008 and rs482hat.  when booting the firmware as burned by your app it gets stuck in a boot loop.   If i provide the exphome with the config files that are from the repository it is able to boot and reads much of the information from the rheem.  the following error is recieved in the logs though.

`[W][component:232]: Component econet took a long time for an operation (55 ms).
[W][component:233]: Components should block for at most 30 ms.
[W][econet:342]:   HOTWATER : UNSUPPORTED
[W][econet:342]:   HRSLOHTR : UNSUPPORTED
[W][econet:342]:   HRSUPHTR : UNSUPPORTED
[W][econet:342]:   POWRWATT : UNSUPPORTED
[W][econet:342]:   VACATIME : UNSUPPORTED
[W][component:232]: Component econet took a long time for an operation (83 ms).
[W][component:233]: Components should block for at most 30 ms.`

modle is 	HB50  as reported by HA.

justbruce, 2024-03-29:
> I jsut completed assembally of the c008 and rs482hat. when booting the firmware as burned by your app it gets stuck in a boot loop. If i provide the exphome with the config files that are from the [...]

The 'UNSUPPORTED' messages are just an indication your device does not report those items.  Do you have an Econet thermostat?

The 'took a long time' messages are also somewhat normal, in my experience.

Phalen, 2024-03-30:
no this is directly from the rheem hot water heatpump.

justbruce, 2024-03-30:
> no this is directly from the rheem hot water heatpump.

Many, if not most of those UNSUPPORTED messages are likely due to the missing Econet thermostat.

tronikos, 2024-03-30:
The UNSUPPORTED ones are expected for some models. Not all models support all sensors. You can either ignore them or manually remove them. In your config add:

```yaml
sensor:
  - id: !remove hot_water
  - id: !remove hours_lower_heater
  - id: !remove hours_upper_heater
  - id: !remove power
number:
  - id: !remove vacatime
```

The "Component econet took a long time for an operation" are also normal. You can ignore them.

The boot loop is potentially a real issue.

brianf21, 2024-03-30:
I'm having the same issue with the same device, boot loop.

justbruce, 2024-03-31:
> I'm having the same issue with the same device, boot loop.
> 
> [esp-web-tools-logs.txt](https://github.com/esphome-econet/esphome-econet/files/14813392/esp-web-tools-logs.txt)

Looks like the mDNS service isn't able to register an event handler.

 @barndawgie  - Any chance this is an upstream issue?

Phalen, 2024-03-31:
> I'm having the same issue with the same device, boot loop.
> 
> [esp-web-tools-logs.txt](https://github.com/esphome-econet/esphome-econet/files/14813392/esp-web-tools-logs.txt)

that looks like the errormessage i was getting.

barndawgie, 2024-03-31:
We have zero code related to this - web tools, improv, etc... are all ESPHome functionality.

brianf21, 2024-03-31:
@barndawgie Thank you. Can you give me the link to submit issues to the upstream project?

justbruce, 2024-03-31:
> @barndawgie Thank you. Can you give me the link to submit issues to the upstream project?

https://github.com/esphome/issues

barndawgie, 2024-03-31:
It's certainly possible this is caused by some issue with out code but I have no idea what or why it's recently started happening.

Issues can be filed here: https://github.com/esphome/esphome/issues/430

barndawgie, 2024-03-31:
You could also ask in the ESPHome Discord: https://discord.gg/6EeFkuytT2

Phalen, 2024-04-01:
Fust FYI  when i copied the 3 files to configure the device it compliled from esphome console.  econet_electric_tank_water_heater.yaml econet_base.yaml waterheater.yaml

brianf21, 2024-04-05:
> Fust FYI when i copied the 3 files to configure the device it compliled from esphome console. econet_electric_tank_water_heater.yaml econet_base.yaml waterheater.yaml

Did you compile each file individually?

Phalen, 2024-04-05:
I am new to esphome so I probably did not do it correctly considering your question. 

below is my main yaml for the device. I suspect it is the wrong way to do it but unfortunatly I could not figure out any better way to do it witht he documentation. i created the other 2 files as esphome asked for them when compiling the installer.  I chopped off the last few lines as they had sensitive info. 

`substitutions:
  name: "waterheater"
  friendly_name: "Water Heater"
  device_description: "Rheem Heat Pump Water Heater"
  board: m5stack-atom

packages:
  econet: !include econet_electric_tank_water_heater.yaml

econet:
  dst_address: 0x1280

climate:
  - id: !extend econet_climate
    custom_presets:
      0: "Off"
      1: "Eco Mode"
      2: "Heat Pump"
      3: "High Demand"
      4: "Electric"
      # Vacation preset doesn't seem to have any effect
      # 5: "Vacation"

sensor:
  - platform: econet
    name: "Ambient Temperature"
    id: ambient_temp
    sensor_datapoint: AMBIENTT
    unit_of_measurement: "°F"
    accuracy_decimals: 1
    device_class: "temperature"
    state_class: "measurement"
    entity_category: "diagnostic"
  - platform: econet
    name: "Evaporator Temperature"
    id: evaporator_temp
    sensor_datapoint: EVAPTEMP
    unit_of_measurement: "°F"
    accuracy_decimals: 1
    device_class: "temperature"
    state_class: "measurement"
    entity_category: "diagnostic"
  - platform: econet
    name: "Suction Temperature"
    id: suction_temp
    sensor_datapoint: SUCTIONT
    unit_of_measurement: "°F"
    accuracy_decimals: 1
    device_class: "temperature"
    state_class: "measurement"
    entity_category: "diagnostic"
  - platform: econet
    name: "Discharge Temperature"
    id: discharge_temp
    sensor_datapoint: DISCTEMP
    unit_of_measurement: "°F"
    accuracy_decimals: 1
    device_class: "temperature"
    state_class: "measurement"
    entity_category: "diagnostic"

binary_sensor:
  - platform: econet
    name: "Compressor"
    id: compressor
    sensor_datapoint: COMP_RLY
    device_class: "running"

text_sensor:
  - platform: econet
    name: "Fan Speed"
    id: fan_speed
    sensor_datapoint: FAN_CTRL
    icon: "mdi:fan"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:`

Sabrown365, 2024-04-11:
I'm having the same issue with mDNS init failed on the same hardware, using the heatpump water heater version to flash the m5stack atom lite. 

@Phalen were you able to resolve your issue?

Is there an earlier version that is known working I can try and flash?

Here's my logs:
[esp-web-tools-logs.txt](https://github.com/esphome-econet/esphome-econet/files/14951864/esp-web-tools-logs.txt)

justbruce, 2024-04-13:
Some research points to this being a problem with the TCP/ip semaphore stack - any chance anything we are doing is trying to communicate before wifi is available?  Does the Atom Lite take a particularly long time to setup the wireless functions?

[specifically looking at the error around `tcpip_send_msg_wait_sem` ]

brianf21, 2024-04-17:
> I had success yesterday using the ESP home dashboard from home assistant to first install the latest ESP home onto the m5 stack, which resulted in no boot looping error. Then installing the yaml from build-yaml/econet-hpwh-esp32.yaml, I did not see the same mDNS boot looping error installing via this method. I used the ESP home dashboard from home assistant to also move over the secrets.yaml.
> 
> So maybe there is a issue using the web page installation method of ESP home econet github.io? That was the only thing I did differently.
> 
> But as of currently everything is running great and all device metadata and control and dashboard data is reporting correctly.

I can't get it to work. I created the bin file, installed it, and validated it fine. I can see that the wifi and password are there, but I keep getting this error.

[W][wifi:152]: Warning set: unspecified
[W][wifi:531]: No matching network found!
[W][econet:193]: New read request while waiting response for previous read request
[W][wifi:531]: No matching network found!
[W][wifi:531]: No matching network found!
[W][econet:193]: New read request while waiting response for previous read request
[W][econet:193]: New read request while waiting response for previous read request
[W][wifi:531]: No matching network found!
[W][econet:193]: New read request while waiting response for previous read request
[W][wifi:531]: No matching network found!
[W][wifi:640]: WiFi network can not be found anymore.
[W][wifi_esp32:479]: Event: Disconnected ssid='**_REDACTED_**' reason='Probe Request Unsuccessful'
[W][wifi_esp32:307]: esp_wifi_connect failed! 12298
[E][wifi:349]: wifi_sta_connect_ failed!

But my SSID and password are correct when I validate the yaml from the build.

justbruce, 2024-04-18:
> I can't get it to work. I created the bin file, installed it, and validated it fine. I can see that the wifi and password are there, but I keep getting this error.
>
> [W][wifi:152]: Warning set: [...]

is there a space in your wifi name?  In other implementations, microcontrollers have had problems with SSIDs that include a space.

  A long shot, but I thought worth asking

Phalen, 2024-04-18:
> @Phalen were you able to resolve your issue?
> 
> Is there an earlier version that is known working I can try and flash?

That was the first time i tried this project as i just got my stack.  

I will be trying with the latest esphome that was just pushed to home assistant the other day. I will also try the latest release shortly. 
i might have updated it once since starting on the stack i belive 3.1 was where i stated. 

Installed version
2024.3.2
Latest version
2024.4.0

@Sabrown365  any thing iot or possibly linux related i dont use special char or spaces with. short and simple where possible. i learned a painfull leason related to special charictors in passwords on linux some years ago as some backend systems may mangle things due to speical char's used in programing and leave you with several different or invalid passwords in an intigrated system.

brianf21, 2024-04-18:
@Phalen Yes, my wifi SSID does have white space, but in the yaml file it is enclosed in double quotes so it should be interpreted as a printable character. But it is worth a shot, I will rename and try again. Thanks for your help.

githubrdonefun, 2024-04-19:
@Sabrown365 Thanks for sharing this solution.

I wanted to add that I also ran into this issue.  I ended up adding it via Home Assistant, which I got to work.  I'm new to ESPHome and this was my first project to learn it.  I was able to solve my problem by finding this thread, but given the number of people reporting this issue, it seems like there's a problem with the website install versus doing it in Home Assistant.

tronikos, 2024-04-20:
I believe it boot loops because WiFi isn't configured. I just tried flashing an ESP32 from web install and I wasn't asked to configure WiFi. 

https://github.com/esphome-econet/esphome-econet/pull/375 is a suspect PR that could have broken this but I verified the improv_serial component is added:
[image]

https://github.com/esphome-econet/esphome-econet/pull/368 is another suspect PR but the ESPHome documentation doesn't mention anything that these components are required. If those components are needed we should add them to the automatic builds similarly to https://github.com/esphome-econet/esphome-econet/pull/375

@barndawgie could you please take a look?

Ylianst, 2024-05-16:
I just updated to ESP-Home 2024.5.0, updated my M5Stack-ATOM and same here, my device is now in a crash loop. I pushed a default empty ESP-Home firmware and the device works again, but when I push the Heat Pump firmware back it goes into a crash loop again.

tronikos, 2024-05-16:
Fixed. It was https://github.com/esphome-econet/esphome-econet/pull/368 that caused this.

PR #387 (2024-05-16): Revert #375
Revert #375 to see if this will resolve #379
PR #395 (2024-05-16): Revert "Remove captive_portal"
Reverts esphome-econet/esphome-econet#368
to see if it will resolve #379
closed 2024-05-16 by tronikos