esphome-econet/esphome-econet #383: Attributes are not getting updated but logs are showing data getting read in

Skullitor13, 2024-04-01:
I have set this up and am seeing data show up in the logs when I turn on verbose logging, but the entities in home assistant is not showing up.

I am getting these lines as well in the logs:

[17:47:22][W][component:232]: Component econet took a long time for an operation (361 ms).
[17:47:22][W][component:233]: Components should block for at most 30 ms.

What other information would be useful?

Skullitor13, 2024-04-02:
[logs_econet-hpwh_logs(1).txt](https://github.com/esphome-econet/esphome-econet/files/14829871/logs_econet-hpwh_logs.1.txt)
Here is about a minute of logs.

With these settings 

logger:
  level: VERY_VERBOSE
 logs:
    scheduler: DEBUG
    component: DEBUG

justbruce, 2024-04-02:
Have you tried deleting the entity and re-adding it?

  Also, what version of Home Assistant?

  The logs look fairly nominal, so I think there is something wrong with the esphome-HAOS connection.

Skullitor13, 2024-04-02:
I deleted the entity and re added it.  That fixed the issue.  Thanks for the help.

Just an FYI I am on the newest version 2024.3.3.

closed 2024-04-02 by Skullitor13