esphome-econet/esphome-econet #388: Rheem Automatic Shut-Off Valve

faultymonk, 2024-04-30:
Various Rheem waterheater models come with automatic leak detection and an [automatic shutoff valve](https://parts.rheem.com/product/RPD-AP20180). HA's EcoNet integration show it as a ["shutoff_valve" binary sensor](https://github.com/home-assistant/core/blob/822646749d32f346ffbb6c7ae5ebae06564fcfa0/homeassistant/components/econet/binary_sensor.py#L20-L24).

I searched through the issues and discussions here but did not seem to be able to find any mention of being able to report on the status of the shutoff valve. Am wondering if this is info we can also obtain through the debug port.

justbruce, 2024-04-30:
Is the valve visible in the Econet Thermostat or in the App?  if so, then there is likely a datapoint that one would just need to find the name of.......

faultymonk, 2024-04-30:
> Is the valve visible in the Econet Thermostat or in the App? if so, then there is likely a datapoint that one would just need to find the name of.......

From the panel, there are a couple of menu items related to it. First is [SOV installed](https://rmc-cdn.s3.amazonaws.com/media/uploads/iat/sites/36/2020/04/AP21681-UseAndCare.pdf) (pg. 19) in the normal menus:

[image]

In diagnostic mode, there is an `SO` or `Shut Off Vavle` [sic] menu where the valve status can be reported.  

[image]

The status is not reported via Rheem's app, but is reported via HA's [econet integration](https://github.com/home-assistant/core/issues/116262).

justbruce, 2024-05-01:
I'd use the HA Services page and the HA Events listening page/section and try SOVEXIST as your datapoint via `ESPHome: econet_hvac_read_from_address` and see if anything comes back.  Basically making an educated guess here based on the known datapoint HRVEXIST.

Thems my thoughts

faultymonk, 2024-05-01:
> I'd use the HA Services page and the HA Events listening page/section and try SOVEXIST as your datapoint via `ESPHome: econet_hvac_read` and see if anything comes back. Basically making an educated guess here based on the known datapoint HRVEXIST.

I just tried a `sensor_datapoint: SOVEXIST` and got:

`[22:58:13][W][econet:342]:   SOVEXIST : UNSUPPORTED
`

faultymonk, 2024-05-01:
holy cow. On a lark, I tried:

```
binary_sensor:
  - platform: econet
    name: "Shut Off Valve"
    id: shutoff_valve
    sensor_datapoint: SHUTOFFV
    device_class: "opening"
```

No error in the logs about the datapoint being unsupported, and:

[image]

tronikos, 2024-05-01:
`UNSUPPORTED ` means this is an invalid datapoint. See https://github.com/esphome-econet/esphome-econet/wiki/Objects for an example response of a valid datapoint.

You will have to keep trying different strings of up to 8 characters to see if you get lucky. I guess you could write an automation in HA that listens to the relevant event and exports to a file or even a sheet and keep trying different datapoints in a loop.

tronikos, 2024-05-01:
Great guess!
To include it here I see the following options:
1. Include it in the regular config. Cons: Users without a shut-off valve will see unknown for this.
2. Include it in the regular config but with `disabled_by_default: true`. Cons: Users with shut-off valve will have to manually enable it. Users without shut-off valve won't see it by default but ESP will keep requesting this datapoint and we have already seen issues especially on ESP8266 when requesting too many datapoints.
3. Put it in a file and require users to manually include that package similar to what we do for HVAC, [example](https://github.com/esphome-econet/esphome-econet/blob/7dcbfefb60d5b7ccd5724b2ba8124050dd96790d/build-yaml/econet-hvac_air_handler-esp32.yaml#L14-L17). Con: Users with shut-off valve will have to manually include the file and compile the firmware
4. Create variant configs for all water heater types to include the shut-off valve. Con: The configs and the prebuilt binaries are already out of hand...

I recommend going with 3. @faultymonk do you want to prepare a PR? Follow the HVAC example I linked to. Make sure you add an icon. BTW, can you modify this datapoint? If yes, wouldn't `switch` be better?

faultymonk, 2024-05-01:
> `UNSUPPORTED ` means this is an invalid datapoint. See https://github.com/esphome-econet/esphome-econet/wiki/Objects for an example response of a valid datapoint.

Thanks for the link to the BACnet documentation. There is a `SHUTSTAT` item with the description "Shut-off Valve State". However, when I tried it, I got back an `unsupported` message.

I decided to google "econet rheem SHUTOFFV" and ended up finding [hubitat rheem econet post](https://community.hubitat.com/t/rheem-econet-integration-maintained-by-kris-linquist/116913/30) where apparently you can connect to the waterheater locally and get an MQTT response that contains (a bunch of interesting) datapoint names.

The ones that look obvious are:
- `SHUTOFFV`: shut-off valve state?
- `SHUTOVER`: shut-off valve override?
- `SHUTOPEN`: shut-off valve open?
- `SHUTCLOS`: shut-off valve close?
```

The interesting thing to note is that the shutoff valve can be controlled via the panel & app. When a water leak is detected, it automatically shuts off. However, this can be overridden by the panel & app (for 24 hrs by a long press to dismiss the alarm) and re-open the valve.

I'll poke around the datapoints a bit more and prep a PR once I'm a bit more sure what's what.

tronikos, 2024-05-01:
Nice find! I opened https://github.com/esphome-econet/esphome-econet/issues/390 to expose some of the other datapoints mentioned in that post.

I believe you also want to expose LSDETECT. I assume it's for the leak sensor.

tronikos, 2024-05-23:
@faultymonk any update?

faultymonk, 2024-05-30:
Sorry for the radio silence. TL;DR - I wouldn't be comfortable unleashing this upon others until I do more work to validate that it won't do something unexpected.

I saw the post from https://github.com/esphome-econet/esphome-econet/issues/390#issuecomment-2136977797 from @Daniel-dev22  as well:

```
"LSDETECT": 1,
"SHUTOFFV": 1,
"SHUTOPEN": 1,
"SHUTCLOS": 0,
```

AFAICT, `SHUTOFFV` appears to be whether the automatic shutoff valve is installed.

`SHUTOPEN` and `SHUTCLOS` are both writable, and you can open and close the valve. I believe they have to be toggled together for things to work. I was poking at them and ended up causing alarms where it complained about invalid valve input states (something like that).  It also alarmed intermittently with something about self-test and not being able to close the valves, but actually left the valve closed (let's just say there were some very unpleased members of the household...). Ultimately I ended up having to power-off the water heater to get this behavior to stop and had second thoughts about whether exposing this would be a good idea...

That and I also keep running into ["New read request while waiting response for previous read request"](https://github.com/tronikos/esphome-econet/blob/351000677a263292f71bbf9398434a58c0be2ada/components/econet/econet.cpp#L193) while trying to add some datapoints...

justbruce, 2025-03-24:
> That and I also keep running into ["New read request while waiting response for previous read request"](https://github.com/tronikos/esphome-econet/blob/351000677a263292f71bbf9398434a58c0be2ada/components/econet/econet.cpp#L193) while trying to add some datapoints...

You may need to add a `request_mod:` to your yaml.  I've noticed that it seems to be possible to overload the little cores in the ESP32, which splitting things into request_mods seems to help with