esphome-econet/esphome-econet #389: Constant diconnections

cjueden, 2024-04-30:
Hi everyone, I'm having an issue with constant disconnections on the M5 with the R485 attachment. 
Does anyone have any suggestions on how to stop this? Attached is a screen shot of the disconnections. I can get any logs needed if someone can point me in the right direction. Thank you in advanced. 
[image]

tronikos, 2024-05-01:
This is a common ESPHome issue, see https://esphome.io/guides/faq.html#my-node-keeps-reconnecting-randomly
If you search on https://community.home-assistant.io/ and https://github.com/esphome/issues/issues you will find many such issues where users see disconnects and unavailable sensors. Try the workarounds mentioned in the FAQ and the many forum posts. Since this isn't an issue of this library I'm closing it. But please update us when you find a workaround that improves this.

closed 2024-05-01 by tronikos

cjueden, 2024-05-01:
Ok I'll look into those. Thank you for taking the time to respond.