esphome-econet/esphome-econet #390: Expose more data for water heaters

tronikos, 2024-05-01:
https://github.com/esphome-econet/esphome-econet/issues/388#issuecomment-2088079342 mentioned https://community.hubitat.com/t/rheem-econet-integration-maintained-by-kris-linquist/116913/30
That includes the following datapoints that we currently don't expose:

Worth exposing:

```
"TOTALKWH": 72.767707,
"HRS_COMP": 102.196487,
"HRSLOFAN": 36.22084,
"HRSHIFAN": 66.070373,
"UNITTYPE": 4,
```

- TOTALKWH will avoid the need of https://github.com/esphome-econet/esphome-econet/wiki/Monitoring-Power-Consumption-in-kWh
- HRS_COMP, HRSLOFAN, HRSHIFAN are similar to the HRSUPHTR and HRSLOHTR we already expose
- For UNITTYPE I get 4 (50 Gallon)

Datapoints that seem related to shut off valve that will be taken care in https://github.com/esphome-econet/esphome-econet/issues/388

```
"LSDETECT": 0,
"SHUTOFFV": 0,
"SHUTOVER": 0,
"SHUTOPEN": 0,
"SHUTCLOS": 0,
```

Not sure what the following represent:

```
"I_RMSVAL": 0.184961,
"EXACTUAL": 100,
"UPELSIZE": 4.5,
"LOELSIZE": 4.5,
"ANODEA2D": 42405,
"ANODESTS": 2,
"DRESOVER": 0,
```

tronikos, 2024-05-01:
For all the unknown datapoints I get the exact values. The only exception is I_RMSVAL for which I get 0.103594. That might be the current in Ampere.

justbruce, 2024-05-01:
"UPELSIZE": 4.5,
"LOELSIZE": 4.5

I would guess Upper Elecrode Size and Lower Electrode Size

elmoret, 2024-05-02:
I_RMSVAL sounds like AC RMS current. AnodeA2D sounds like anode analog to digital, meaning a digital reading of an analog signal, probably voltage I'd guess.

faultymonk, 2024-05-04:
> * TOTALKWH will avoid the need of https://github.com/esphome-econet/esphome-econet/wiki/Monitoring-Power-Consumption-in-kWh

```
  - platform: econet
    name: "Electric Consumption [kWh]"
    id: electric_consumption_kwh
    sensor_datapoint: TOTALKWH
    unit_of_measurement: "kWh"
    accuracy_decimals: 6
    device_class: "energy"
    state_class: "total_increasing"
```

Wanted to add a note about `TOTALKWH`. I've been testing this, and it matches what the Rheem app reports (mostly). The return value is only updated over some periodic interval, so it can seem like it's under-reporting the value when compared w/ the Riemann sum method until it's updated.

Also, it seems that if certain data points are added or if there are too many, ["New read request while waiting response for previous read request"](https://github.com/tronikos/esphome-econet/blob/351000677a263292f71bbf9398434a58c0be2ada/components/econet/econet.cpp#L193) errors start to appear in the logs.

tronikos, 2024-05-04:
I'll personally continue using the Riemann sum on the POWRWATT mostly because I've calibrated that sensor with:
```yaml
substitutions:
  sensor_power_filters_lambda: return x * 0.91 - 20;
```
But I think there is value exposing TOTALKWH since it simplifies things for average users.

Yes having too many requests slows down the MCU causing symptoms like the one you mentioned. That's why I first want to remove some redundant alarm history sensors in #371 to make room for the new request in #391

justbruce, 2024-05-04:
> Yes having too many requests slows down the MCU causing symptoms like the one you mentioned. That's why I first want to remove some redundant alarm history sensors in #371 to make room for the new request in #391

It seems [from logs I've read from others, as well as my own] that at present, every datapoint is being requested twice, and that REQUEST_ONCE is being ignored - this would certainly increase the load on the MCU.  Might this be a more fruitful problem to chase down before eliminating Alarm History data points?

tronikos, 2024-05-04:
@justbruce can you open an issue with lugs attached?

kaijk, 2024-05-09:
> I'll personally continue using the Riemann sum on the POWRWATT mostly because I've calibrated that sensor with:
> 
> ```yaml
> substitutions:
>   sensor_power_filters_lambda: return x * 0.91 - 20;
> ```
Question: with the use of github Packages and a sparse master file, where does one put this substitution for Watts?
Thx

tronikos, 2024-05-09:
Just add it to the beginning of the file. Example:

```yaml
---
substitutions:
  sensor_power_filters_lambda: return x * 0.91 - 20;

packages:
  esphome-econet.esphome-econet: github://esphome-econet/esphome-econet/build-yaml/econet-hpwh-esp8266.yaml@main

api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Alternatively, it's better to add the following at the end of your configuration:

```yaml
sensor:
  - id: !extend power
    filters:
      - lambda: return x * 0.91 - 20;
```

The advantage of that is you can use [calibrate_linear](https://esphome.io/components/sensor/index.html#calibrate-linear) or [calibrate_polynomial](https://esphome.io/components/sensor/index.html#calibrate-polynomial).

tronikos, 2024-05-17:
The remaining ones are now:

```
"UPELSIZE": 4.5,
"LOELSIZE": 4.5,
"ANODEA2D": 42405,
"ANODESTS": 2,
"DRESOVER": 0,
```

tronikos, 2024-05-29:
@Daniel-dev22 can you post your findings in #388 ? That one is taking care of adding support for leak detector and shut off valve.

cfelicio, 2024-08-09:
Curious what the expansion valve actually is. I thought at first it was related to the shutoff valve, but it stays at 100% unless the heat pump is running, then it drops. Maybe it's a mechanism to control something on the heat pump?

[image]

barndawgie, 2025-07-11:
`SHUTOFFV` works for me - returns 0/no (I don't have one). So that seems pretty safe to add to the basic water heater config.

barndawgie, 2025-07-11:
This worked great in my config. Not sure how useful they really are, though.

```yaml
binary_sensor:
  - platform: econet
    name: "Leak Sensor Installed"
    id: leak_sensor_installed
    sensor_datapoint: LSDETECT
    entity_category: "diagnostic"
    request_mod: 4
    request_once: true
    icon: "mdi:water-pump"
  - platform: econet
    name: "Shut-Off Valve Installed"
    id: shutoff_valve_installed
    sensor_datapoint: SHUTOFFV
    entity_category: "diagnostic"
    request_mod: 4
    request_once: true
    icon: "mdi:valve"
```

justbruce, 2025-07-13:
> This worked great in my config. Not sure how useful they really are, though.

Maybe set them as disabled_by_default and let users enable them if they have use for them in automations or otherwise?