esphome-econet/esphome-econet #401: Add 'hvac_write_float_from_address' HA Service

justbruce, 2024-05-28:
The current 'hvac_write_float' defaults to the 0x340 FROM address [the wifi module].  Many writes are not accepted/immediately overwritten by the thermostat from this address [for example FILTUSED and FILT_CHG].  

  For testing reasons [at present], it would be helpful to be able to write a float value with a FROM address of the Furnace, or the ODU.   

Would it be possible to get an HA Service like this?  To more effectively explore the protocol, at least?

tronikos, 2024-05-29:
#402 adds such as a service.

Can you verify it works before merging it?

You can use it by replacing the package in your yaml with:

**Don't forget to update the file below to match your device**

```yaml
packages:
  econet:
    url: https://github.com/tronikos/esphome-econet
    ref: write_float_from_address
    file: build-yaml/econet-hpwh-esp8266.yaml
    refresh: 0s
```

tronikos, 2024-06-08:
@justbruce did you get a chance trying out #402 ?

justbruce, 2024-06-08:
Not yet.  Life has been busy otherwise, sorry.

PR #402 (2024-06-21): Add write_float_from_address and write_enum_from_address
Fixes #401
closed 2024-06-21 by tronikos