esphome-econet/esphome-econet #403: every datapoint is being requested twice, and REQUEST_ONCE is being ignored

faultymonk, 2024-05-30:
> Yes having too many requests slows down the MCU causing symptoms like the one you mentioned. That's why I first want to remove some redundant alarm history sensors in #371 to make room for the new request in #391

It seems [from logs I've read from others, as well as my own] that at present, every datapoint is being requested twice, and that REQUEST_ONCE is being ignored - this would certainly increase the load on the MCU.  Might this be a more fruitful problem to chase down before eliminating Alarm History data points?

_Originally posted by @justbruce in https://github.com/esphome-econet/esphome-econet/issues/390#issuecomment-2094204913_

faultymonk, 2024-05-30:
Attaching excerpted debug logs. PLMK if this isn't sufficient and there are specific bits I should look for and include.

[debug_log_excerpt.txt](https://github.com/esphome-econet/esphome-econet/files/15494502/debug_log_excerpt.txt)

tronikos, 2024-05-31:
In the logs I don't see any duplicate request. Can you point me to specific lines?

tronikos, 2024-05-31:
And do you have an example that request once is ignored?

tronikos, 2024-06-08:
Reopen once you have logs showing this happening.

closed 2024-06-08 by tronikos