esphome-econet/esphome-econet #409: Enable esphome-econet to use new `http-request` OTA capability of ESPHome

barndawgie, 2024-06-20:
We should consider updating esphome-econet to use the newly introduced HTTP-request OTA updates in ESPHome:
https://esphome.io/components/ota_http_request

tronikos, 2024-06-20:
What are the benefits? Is the idea to point to the firmware files we built? It's not clear from the documentation whether the update happens fully automatically or the user has to initiate it. My understanding is it allows users to update without needing to setup the ESPHome addon. I don't feel very comfortable auto updating the devices. Apart from breaking changes there are security concerns, e.g. one of our github accounts gets compromised and the bad actor pushes a new malicious firmware. On top of that, it seems it requires HTTP Request component which seems large, especially for ESP8266, for which the documentation says you might have to disable SSL due to memory constraints which has even more security implications.

barndawgie, 2024-06-21:
I'm going to try setting it up on my device locally and see how it works. I need #410 to do that as it will cause our build workflow to generate the necessary binaries and checksums as part of our release process.