esphome-econet/esphome-econet #412: No data in the logs

rsmanning, 2024-07-06:
I'm pretty new to all this but I followed AAlliieennX's instructions exactly, purchased the recommended hardware, flashed from the github page with "esphome-econet.esphome-econet: github://esphome-econet/esphome-econet/build-yaml/econet-hpwh-esp32.yaml@main" in the packages yaml.  Its a heatpump type water heater so I used the firmware option.

I'm not getting any data back from mine in the logs and triple checked the wires are in the write location and Ohm tested from the phone pins to the screw heads on the phoenix connector that the connection is good.  Its an Econet controller with a 6pin phone jack on the side.

rsmanning, 2024-07-06:
Its a Richmond, not sure if that makes a difference but from what I read Rheem makes Richmond and the part numbers cross over.  This is a model: 10E80-HP4D.

barndawgie, 2024-07-06:
Are you running the ESPHome dashboard? If so, adopt the device and you can try adding this to your config:

```
substitutions:
  logger_level: INFO
```

Then share your logs here.

rsmanning, 2024-07-06:
My bad, you can delete this.  looking at pictures on the github page for connecting wires I thought I had them right.  Looked like red to red and green to green on the atom (thought that was strange with it labeled 12v) but then I read the pin out text and switched it to A and B, its working now.

closed 2024-07-06 by tronikos