esphome-econet/esphome-econet #416: Seems like it's set up, but the only data I'm getting is wifi strength.

Obioban, 2024-07-16:


Obioban, 2024-07-16:
Log file:
`INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/econet-hpwh.yaml...
INFO Starting log output from 192.168.1.86 using esphome API
INFO Successfully connected to econet-hpwh @ 192.168.1.86 in 0.060s
INFO Successful handshake with econet-hpwh @ 192.168.1.86 in 0.093s
[12:02:50][W][econet:195]: New read request while waiting response for previous read request
[12:02:56][W][econet:195]: New read request while waiting response for previous read request
[12:03:02][W][econet:195]: New read request while waiting response for previous read request
[12:03:08][W][econet:195]: New read request while waiting response for previous read request
[12:03:08][W][component:237]: Component econet took a long time for an operation (58 ms).
[12:03:08][W][component:238]: Components should block for at most 30 ms.
[12:03:20][W][econet:195]: New read request while waiting response for previous read request
[12:03:26][W][econet:195]: New read request while waiting response for previous read request
[12:03:32][W][econet:195]: New read request while waiting response for previous read request
[12:03:38][W][econet:195]: New read request while waiting response for previous read request
[12:03:38][W][component:237]: Component econet took a long time for an operation (59 ms).
[12:03:38][W][component:238]: Components should block for at most 30 ms.`

Obioban, 2024-07-16:
Thoughts on how to fix this?

I'm using a Olimex ESP32-POE-ISO and MOD-RS485, with the two flow control pins soldered together. 

Currently using wifi to not mix the pot too much, but my goal is to get this to be PoE

Obioban, 2024-07-16:
[image]
[image]

elmoret, 2024-07-17:
Would be useful to see the other end of those cables...

barndawgie, 2024-07-17:
This is almost always caused by mis-wiring. Please share a clear picture of both ends of the wire.

Obioban, 2024-07-18:
MOD-RS485:

[image]
[image]

End that plugs into the water heater:

[image]
[image]

barndawgie, 2024-07-18:
Great pics! Those do look right. Next thing to check might be that you have the correct GPIO pins configured. Our config includes the mappings for our recommended hardware (the m5Stack Atom Lite with m5Stack RS485 interface:

```yaml
substitutions:
  tx_pin: GPIO19
  rx_pin: GPIO22
``

You'll likely need to override those to the correct GPIO's for your ESP32/ESP8266 hardware.

elmoret, 2024-07-18:
You'll need to refer to your devices' schematics:

https://github.com/OLIMEX/ESP32-POE-ISO/blob/master/HARDWARE/ESP32-PoE-ISO-Rev.B/ESP32-PoE-ISO_Rev_B.pdf
https://www.olimex.com/Products/Modules/Interface/MOD-RS485/open-source-hardware

and the setup instructions:
https://github.com/esphome-econet/esphome-econet/wiki/Schematics

So GPIO4 and GPIO36 are your TX/RX pins:

```
substitutions:
  tx_pin: GPIO4
  rx_pin: GPIO5
  platform: esp8266
```

And you need to tell it what your using for your flow control pin. I can't tell what you're doing with that yellow wire, hopefully bridging DE and RE. Pick a pin in the UEXT connector that results from your bridge, and define it in your config file. Mabe GPIO14?

```
econet:
  flow_control_pin: GPIO14
```

closed 2024-08-31 by barndawgie