esphome-econet/esphome-econet #427: Read requests timing out

ericmagaha, 2024-11-19:
Sooo....setup an M5Stack Atom S3 Lite with the A131 RS485 converter.  Successfully flashed the unit using econet-hpwh-esp32s3.yaml from the build-yaml directory i.e. copied it into config and added a secret.yaml etc for wifi.  I double checked the wiring and it is wired correctly.

Here is what I did:

1.  Flashed the unit using esphome from the command line in a Python venv.  Ran "esphome run econet-hpwh-esp32s3.yaml" and flashed the unit via direct serial.
2. Plugged the unit into the Rheem heat pump water heater.
3. Ran "esphome config dashboard" and hit the URL on 6502.  I can see the device but it only gives me options for "edit" and "logs".  I can successfully view the logs.
4. It shows the following:
INFO ESPHome 2024.10.3
INFO Reading configuration config/econet-hpwh-esp32s3.yaml...
INFO Starting log output from 192.168.1.106 using esphome API
INFO Successfully connected to econet-hpwh @ 192.168.1.106 in 0.031s
INFO Successful handshake with econet-hpwh @ 192.168.1.106 in 0.017s
[18:27:00][W][econet:195]: New read request while waiting response for previous read request
[18:27:07][W][econet:195]: New read request while waiting response for previous read request
[18:27:13][W][econet:195]: New read request while waiting response for previous read request
[18:27:19][W][econet:195]: New read request while waiting response for previous read request
[18:27:25][W][econet:195]: New read request while waiting response for previous read request
[18:27:25][W][component:237]: Component econet took a long time for an operation (53 ms).
[18:27:25][W][component:238]: Components should block for at most 30 ms.

The econet-hpwh-esp32s3.yaml specifies GPIO6 for tx and GPIO5 for rx.  I used that initially.  Then I changed it to GPIO2 and GPIO1 as specified under the "Proven ESP Hardware" section and re-flashed it over wifi.  Same thing.

Any help?

Thanks.

barndawgie, 2024-11-20:
Those messages are pretty common, especially with more verbose logging enabled. Are all of your sensors working? If so, safe to ignore them.

ericmagaha, 2024-11-20:
Thanks for the response.

1.  How do I view the sensor data?  I do not use Home Assistant.  I was under the impression I could view it using the Esphome Dashboard.  This is what I see in Esphome Dashboard.
[image]

2.  Can I edit the source code to change the logging level?

Thanks.

tronikos, 2024-11-20:
https://github.com/esphome-econet/esphome-econet/discussions/426

closed 2024-11-20 by tronikos

ericmagaha, 2024-11-20:
Ah right, did watch that video but I am also not using MQTT.

Actually found what I needed by adding a web server to the yaml.
web_server:
  port: 80
  version: 3
  local: true

I can hit the root and see all the data in the HTML table.  And it looks like there is a REST API that I can use, which is perfect.