esphome-econet/esphome-econet #429: Failed config - Expected visual_temperature with unit, got 1e-15.     temperature_step: 1.0f

boctok, 2024-11-21:
Tried updating to latest ESPHome Firmware: 2024.11.0

Got this:

INFO ESPHome 2024.11.0
INFO Reading configuration /config/esphome/econet-etwh.yaml...
Failed config

climate.econet: [source /data/packages/e86e50df/econet_electric_tank_water_heater.yaml:15]
  platform: econet
  id: econet_climate
  name: None
  visual: 
    min_temperature: 43.3333
    max_temperature: 60
    
    Expected visual_temperature with unit, got 1e-15.
    temperature_step: 1.0f
  current_temperature_datapoint: UPHTRTMP
  target_temperature_datapoint: WHTRSETP
  mode_datapoint: WHTRENAB
  modes: 
    0: OFF

I tried reinstalling, but got the same thing

boctok, 2024-11-21:
Thanks! That did the trick.

johncarey70, 2024-11-21:
I have the same issue but using econet_hvac_furnace.yaml with zoning: econet_hvac_zone1.yaml
Since the  zone1.yaml removes the econet_climate, how do i extend it to include that fix?

imrj, 2024-11-21:
> I have the same issue but using econet_hvac_furnace.yaml with zoning: econet_hvac_zone1.yaml Since the zone1.yaml removes the econet_climate, how do i extend it to include that fix?

just try adding it anyways see if it fixes it?

johncarey70, 2024-11-21:
Yea I did try it.
Source for removal of ID 'econet_climate' was not found.

tronikos, 2024-11-21:
I'll send a PR fix soon. Testing that it still compiles with older versions.

samsonusmc, 2024-11-21:
Any chance you can do the same for the [econet_heatpump_water_heater.yaml](https://github.com/esphome-econet/esphome-econet/blob/main/econet_heatpump_water_heater.yaml) also?

tronikos, 2024-11-21:
No change needed there. It includes econet_electric_tank_water_heater.yaml

samsonusmc, 2024-11-21:
hmmmm...I get the error on install
I am trying to figure out why HA is pulling an old file even after I purge the build files.

`INFO ESPHome 2024.11.0
INFO Reading configuration /config/esphome/esp-econet.yaml...
Failed config

climate.econet: [source /data/packages/e86e50df/econet_electric_tank_water_heater.yaml:15]
  platform: econet
  id: econet_climate
  name: None
  visual: 
    min_temperature: 43.3333
    max_temperature: 60
    
    Expected visual_temperature with unit, got 1e-15.
    temperature_step: 1.0f
  current_temperature_datapoint: UPHTRTMP
  target_temperature_datapoint: WHTRSETP
  mode_datapoint: WHTRENAB
  modes: 
    0: OFF`

tronikos, 2024-11-21:
What's your full config?

samsonusmc, 2024-11-21:
`wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Econet Fallback Hotspot"
    password: "ZB15fLEFoDVr"

    
substitutions:
  platform: esp32
  board: m5stack-atom
  tx_pin: GPIO19
  rx_pin: GPIO22

packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    file: econet_heatpump_water_heater.yaml

web_server:
  port: 80

climate:
  - id: !extend econet_climate
    visual:
      temperature_step: 0.55555  # Equivalent of 1F in C

captive_portal:
    
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: master_bed_ir ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: master_bed_ir IP
    ssid:
      name: master_bed_ir SSID
    bssid:
      name: master_bed_ir BSSID

dashboard_import:
  package_import_url: github://esphome-econet/esphome-econet/build-yaml/${name}-${platform}.yaml@${github_ref}
  import_full_config: false
`

samsonusmc, 2024-11-21:
Sorry.. the above has the workround.. w/out it, fails to build in ESP Home in HA

tronikos, 2024-11-21:
Add `refresh: 0s` so that it becomes:

```
packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    file: econet_heatpump_water_heater.yaml
    refresh: 0s
```

And remove the `climate:` part. That workaround is no longer needed.

samsonusmc, 2024-11-22:
This addressed it! Thank you!!