esphome-econet/esphome-econet #435: 404 error installing HPWH firmware

lsnider, 2024-12-03:
We are getting a 404 error when trying to install the heat pump firmware.   Here is a screen shot of the error message.  We've tried this in safari and chrome on a mac and edge on windows and all get the same error message.  We also verified we can connect and see the logs of the device.  Just seems to be a broken link or something for Heatpump Water Heater firmware maybe?

[image]

barndawgie, 2024-12-03:
Thanks! I will try to look into soon; there have been changes made to a bunch of stuff here, so I suspect there is something I need to adapt the tools to.

In the meantime, you should still be able to install via the directions https://github.com/esphome-econet/esphome-econet/wiki/Initial-ESPHome%E2%80%90econet-Software-Configuration-and-Installation

barndawgie, 2024-12-03:
The site is point you to this Manifest: https://esphome-econet.github.io/esphome-econet/econet-hpwh.json

The Manifest exists and contains the following:
```json
{
  "name": "esphome-econet",
  "new_install_improv_wait_time": 15,
  "new_install_prompt_erase": false,
  "version": "v2.2.2",
  "home_assistant_domain": "esphome",
  "builds": [
    {
      "chipFamily": "ESP32",
      "ota": {
        "path": "econet-hpwh-esp32.ota.bin",
        "md5": "11332eee915f1f3aa9f6055fdbc8d5b4"
      },
      "parts": [
        {
          "path": "econet-hpwh-esp32.factory.bin",
          "offset": 0
        }
      ]
    },
    {
      "chipFamily": "ESP32-S3",
      "ota": {
        "path": "econet-hpwh-esp32s3.ota.bin",
        "md5": "6be0bcd893454d24e96070cd04716cbc"
      },
      "parts": [
        {
          "path": "econet-hpwh-esp32s3.factory.bin",
          "offset": 0
        }
      ]
    },
    {
      "chipFamily": "ESP8266",
      "ota": {
        "path": "econet-hpwh-esp8266.ota.bin",
        "md5": "97be31c17606be874241482753ca5c07"
      },
      "parts": [
        {
          "path": "econet-hpwh-esp8266.factory.bin",
          "offset": 0
        }
      ]
    }
  ]
}
```

barndawgie, 2024-12-03:
I think the issue is that these files are ending up in a different path than they used to.
https://esphome-econet.github.io/esphome-econet/econet-hpwh-esp32.factory.bin returns a 404, while https://esphome-econet.github.io/esphome-econet/econet-hpwh-esp32/econet-hpwh-esp32.factory.bin is the desired package. Just need to figure out where this is getting broken/how it can be fixed. Likely it's in the GitHub action that "Consolidates" the firmwares into a single payload.

lsnider, 2024-12-04:
I meant to come back on and say I was able to download that file and install the yaml file by hand.  And everything worked great!  So yea I bet it's just a broken path.

PR #436 (2024-12-05): Fix #435
Fixing some pathing bugs in the Github Actions that were blocking ESP Web Tools from working to Fix #435.
commit b30e8e34 (2024-12-05): Fix #435 (#436)
closed 2024-12-05 by tronikos