esphome-econet/esphome-econet #438: Temperature sometimes off by one degree when changing from HomeKit (rounding issue?)

jdwhite, 2024-12-16:
ESPHome 2024.11.3
esphome-econet v2.2.4
iOS 18.2

I'm not sure precisely when I started noticing this issue -- within the last month or so, but I have tried earlier versions of both ESPHome and the Econet integration from several months back when I definitely didn't have this issue, and I still see the following behavior. It's possible that iOS 18 is to blame, but please hear me out.

Problem: when setting the temperature of my tankless from HomeKit the temp that's actually getting set on my tankless (verified by the readout on the physical control unit) is _sometimes_ one degree less than requested. I've enabled debugging and here's what happening with various temp requests:

```
[02:11:59][I][app:100]: ESPHome version 2024.11.3 compiled on Dec 16 2024, 02:08:02
[02:11:59][I][app:102]: Project esphome-econet.esphome-econet version v2.2.4
```

**Setting 112F from HomeKit (not working)**
```
[02:12:05][D][climate:011]: 'Tankless Water Heater' - Setting
[02:12:05][D][climate:040]:   Target Temperature: 44.40
[02:12:05][D][econet:533]: Setting datapoint WHTRSETP to 111.919998
```

Note that the Target temp is "44.40" here.  Physical tankless display shows "111". Looks like WHTRSETP is being truncated to an int.

**Setting 112F from HASS climate control (working)**

```
[02:51:04][D][climate:011]: 'Tankless Water Heater' - Setting
[02:51:04][D][climate:040]:   Target Temperature: 44.44
[02:51:04][D][econet:533]: Setting datapoint WHTRSETP to 112.000000
```
Note  that the target temp is "44.44" instead of "44.40" from HomeKit.  Physical display reads "112".

**Setting 130F from HomeKit**
```
[02:17:54][D][climate:011]: 'Tankless Water Heater' - Setting
[02:17:54][D][climate:040]:   Target Temperature: 54.40
[02:17:54][D][econet:533]: Setting datapoint WHTRSETP to 129.919998
```

Note that the Target temp is "54.40" here.  Physical tankless display shows "129". Same behavior as 112F; 129.xxx being truncated to an int?

**Setting 130F from HASS climate control (working)**

```
02:55:45][D][climate:011]: 'Tankless Water Heater' - Setting
[02:55:45][D][climate:040]:   Target Temperature: 54.44
[02:55:45][D][econet:533]: Setting datapoint WHTRSETP to 130.000000
```

Note target temp from HASS is 54.44 and 54.40 from HomeKit. Physical display reads "130".

I no longer have any iOS 17 devices to verify whether or not HomeKit sent different values prior to iOS 18, but regardless it appears that rounding the target temp after converting from C to F and using that value for WHTRSETP would resolve this issue.

I've also noticed that when using HomeKit to set the temp, all the values in the hundredths place show as zero -- as also shown in the examples above.

tronikos, 2024-12-16:
This seems to be an issue in the homekit integration

Which HA version do you have?
2024.12 might have fixed this in https://github.com/home-assistant/core/pull/131973
See https://github.com/home-assistant/core/issues/98578 for some debugging instructions
You will likely need to open an issue there.

melyux, 2024-12-24:
Having this exact problem since updating to 2024.12. Setting to 74 from HomeKit seems to convert to 23.3, then back to Fahrenheit to 73.94, then rounds down to 73 and sets it to 73. Utterly incomprehensible. Hopefully will be fixed by #131973.

barndawgie, 2024-12-24:
I'm going to resolve this, then, since it's sounds like it's either a HomeKit integration or Home-Assistant Core issue.

closed 2024-12-24 by barndawgie