esphome-econet/esphome-econet #464: New model hybrid heat pump not working

jdanders, 2025-03-16:
I have a new heat pump/furnace model and can't get esphome-econet to work. I'm pretty sure that I have the hardware connected correctly because I scoped the RS-485 lines and can see data being transmitted. 

When I plug into the diagnostic port, the little `rx` LED on the RS-485 board lights up faintly, but it does turn on full brightness periodically like the furnace is transmitting. 

I turned on UART debug to see if I was getting any data, and there is some stuff, but it might be mostly noise. I'll unplug it later and see if it still gets the same "input" data.

Here is some example log info.
```
[22:49:41][D][uart_debug:114]: <<< FF:EB:61:33:33:BF:BF:BF:BF:BF:BF:FF:E7:FF:FF:F1:6F:35:3D:17:2D:23:31:EF:FF:FF:F9:63:21:BF:F1:FF:7B:FF:EF:FF:FF:F9:61:33:33:E3:FF:FF:ED:61:33:33:BF:BF:BF:BF:BF:BF:E9:FF:FF:F3:61:33:33:BF:BF:BF:95:FE:F9:F9:7F:FF:E7:FF:FF:C3:FD:FD:FF:FF:59:75:5B:6D:7D:67:41:63:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF
[22:49:41][I][econet:395]: Read 120. ms=112400
[22:49:41][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F9:7F:FF:F9:D7:FF:FF:F3:FF:FF:FF:FF:FF:FF:FF:FF:FF:ED:E5:D3:E7:FF:C5:B7:0F:FE:F7:FF:FF:F9:7F:FF:E7:FF:FF:C3:FD:FD:FF:FF:59:75:5B:6D:7D:67:41:63:A1:29:FF:FF:FF:F9:7F:FF:F7:FF:FF:D7:FF:FF:F3:FF:FF:FF:FF:FF:FF:FF:FF:FF:EF:FD:D1:F5:E7:AD:3F:41:E7:FE:FD:F9:7F:FF:E7:FF:FF:C3:FD:FD:FF:FF:59:75:5B:6D:7D:67:41:63:47:03:F9:7F:FF:FD:D7:FF:FF:F3:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:CB:CF:D1
[22:49:41][I][econet:395]: Read 120. ms=112466
[22:49:41][D][uart_debug:114]: <<< B3:CB:05:75:FF:FE:F9:F9:7F:FF:E7:FF:FF:C3:FD:FD:FF:FF:59:51:41:53:75:5B:59:63:D9:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F9:7F:FF:F9:C7:FF:FF:F3:FF:FF:FF:FF:FF:FF:FF:FF:FF:7D:79:A5:5B:6F:55:6D:A5:9F:99:A5:9F:99:A5:9F:97:FF:FE:F7:FF:FF:F9:7F:FF:E7:FF:FF:C3:FD:FD:FF:FF:59:51:41:53:75:5B:59:63:FF:FF:FF:FF:FF:FF:F9:7F:FF:F7
[22:49:41][D][uart_debug:114]: <<< FF:FF:C7:FF:FF:F3:FF:FF:FF:FF:FF:FF:FF:FF:FF:7D:79:A5:55:63:6D:61:A5:9F:9F:A5:9F:9D:A5:97
[22:49:41][I][econet:162]: >>> 80.00.03.80.00.80.00.03.40.00.0C.00.00.1E.02.01.00.00.46.41.4E.5F.4D.4F.44.45.A2.28 (28)
[22:49:41][I][econet:164]:   Dst Adr : 0x380
[22:49:41][I][econet:165]:   Src Adr : 0x340
[22:49:41][I][econet:166]:   Command : 30
[22:49:41][I][econet:167]:   Data    : 02.01.00.00.46.41.4E.5F.4D.4F.44.45 (12)
[22:49:41][I][econet:182]:   Type    : 2
[22:49:41][I][econet:183]:   PropType: 1
[22:49:41][W][econet:195]: New read request while waiting response for previous read request
[22:49:41][I][econet:200]:   FAN_MODE
[22:49:41][I][econet:395]: Read 59. ms=112600
[22:49:41][D][uart_debug:114]: >>> 80:00:03:80:00:80:00:03:40:00:0C:00:00:1E:02:01:00:00:46:41:4E:5F:4D:4F:44:45:A2:28
[22:49:41][D][uart_debug:114]: <<< 95:FF:FE:FD:F9:7F:FF:E7:FF:FF:C3:FD:FD:FF:FF:59:51:41:53:75:5B:59:63:FF:FF:F9:7F:FF:FD:C7:FF:FF:F3:FF:FF:FF:FF:FF:FF:FF:FF:FF:7D:79:A5:73:5B:63:9D:A5:73:9D:A5:9F:9D:A5:9D:8F:FF
[22:49:42][I][econet:395]: Read 120. ms=113341
[22:49:42][D][uart_debug:114]: <<< FE:FD:F9:E7:FF:FF:C3:FD:FD:FF:FF:6F:51:59:57:7D:57:55:59:FE:F9:FD:FF:F3:FF:FF:FF:FF:FF:79:F1:FF:FF:FF:FF:FF:FF:00:FF:FD:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FD:7C:B1:FF:FF:FF:FF:89:57:FB:80:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:E7:FF:F9
```

This model has built-in bluetooth for installer and diagnostics. I'm afraid maybe they disabled the port or updated something.

Any tips on next steps?

bitfliq, 2025-03-16:
Interesting find.  As @barndawgie will likely chime in - it may be worthwhile to double check the wires or try an alternative configuration.  Can you share what model of furnace/heat pump and thermostat you have?

barndawgie, 2025-03-16:
+1 to what @bitfliq said - so far, all cases of "not getting any data" have been some form of wiring issue: Mis-mapped wires, bad connection, device with no terminating resistor, etc... So I would assume that before thinking they had changed device communication spec.

Can you also please describe your setup? What ESP hardware and RS485 interface are you using?

justbruce, 2025-03-17:
> I have a new heat pump/furnace model and can't get esphome-econet to work. I'm pretty sure that I have the hardware connected correctly because I scoped the RS-485 lines and can see data being [...]

IME, whenever you have a diagnostic LED faintly lighting it is one of 2 things

1. Insufficient supply voltage on the line 
2.  Short to ground [usually through a diode or low-value resistor]

The short to ground seems likely here, as it will often leave what looks like binary pulses which can be interpreted, but due to the low signaling voltage ends up as garbage output

I guess what I'm saying is - check your wiring.  Better yet, post a picture of your wiring and the interface pieces you're using.

jdanders, 2025-03-17:
Thanks all for the quick and helpful responses. I took some more time looking at my hardware and I think the signal integrity is good to the ESP chip. Some replies:

> Can you share what model of furnace/heat pump and thermostat you have?

I'm not sure if this model number helps or not, but it's what the installer provided.
R96V85 Coil: RCFZ6024STANMC (probably not relevent)
RD748 Furnace: R962V0855A21M4C

> Can you also please describe your setup? What ESP hardware and RS485 interface are you using?

I have an ESP32-c3 with the UART on pins 5 rx / 6 tx as recommended for S3 which is what I started with. I can change to 20/21 recommended in the example YAML, but as I understand ESPHome documentation it shouldn't matter for the ESP32. I have the board powered by a micro-USB plug, but I tried built-in USB-C as well to be sure.

The RS485 board has a MAX485ESA chip on it, and appears to be wired TX/TX according to the back. 

> I guess what I'm saying is - check your wiring. Better yet, post a picture of your wiring and the interface pieces you're using.

[image]

[image]

[image]

Hopefully that all looks good.

> whenever you have a diagnostic LED faintly lighting it is one of 2 things

I applied my scope to the board while in the furnace and realized that the faint LED is caused by the furnace putting out a brief periodic pulse while the door was open (which turns the system off). When I pushed the "door closed" button to turn the system on, the LED went full brightness and the scope says the signals look good. 

Here's a capture of data from the furnace from the `RX` pin on the ESP board.

[image]

This looks like a baud rate of 38400 because the period looks to be 260 us. The data decoded by the scope is 0x80,00,04,00,00,80,00,etc which looks good. 

However, in addition to being 38400 baud, the signal is also inverted. The RX pin sits low while idle, where typically in my experience a UART should idle high and drop the signal for the start bit.

Do I need to swap my +/- signals on the RS485? I'll give that a try.

jdanders, 2025-03-18:
Looks like swapping the green and red wires on the RS485 side of things fixed it. Green is in the A+ port and red in the B- port.

This is awesome data! Expect some upcoming issues and/or pull requests as I look through things. I've got some suggestions on organization, and my "Supply Air Temperature" is showing 6,514 °F right now 😄

closed 2025-03-18 by jdanders